Screenly/Anthias #1410: API_URL tries to resolve to the SSL enabled version instead of HTTP because of is_demo_node

ealmonte32, 2020-11-16:
#### Overview of the Issue

API_URL tries to resolve to the SSL enabled version of the URL instead of HTTP because of `is_demo_node` but the RPi is not set up as demo node..

When is_demo_node bool function is commented out, the API_URL properly gets assigned.

Code section that was edited:
```
else:
    SWAGGER_URL = '/api/docs'
    swagger_address = getenv("SWAGGER_HOST", my_ip)

    if settings['use_ssl']: # or is_demo_node:
        API_URL = 'https://{}/api/swagger.json'.format(swagger_address)
    elif LISTEN == '127.0.0.1' or swagger_address != my_ip:
        API_URL = "http://{}/api/swagger.json".format(swagger_address)
    else:
        API_URL = "http://{}:{}/api/swagger.json".format(swagger_address, PORT)
```

#### Reproduction Steps

Without enabling SSL on your Screenly OSE Pi, visit the API url http://ip.of.your.rpi/api/docs/

You should get a:
```
Failed to load spec.
and
Online validator badge showing INVALID
```

[image]

closed 2022-01-09 by stale[bot]

ealmonte32, 2022-11-17:
Need to solve this so want to open up this ticket again..

config has use_ssl as off, and the Pi is not as demo_node, so swagger should be trying to connect as "https"..

[image]

vpetersson, 2022-11-17:
Good find.

ealmonte32, 2022-11-17:
Current page when pressing API docs link:

[image]

If manually change address to use HTTP, swagger works:
[image]