Screenly/Anthias #1411: Failed to install the experimental version on raspberry Pie 4 in run docker compose

wmysummer, 2020-11-19:
#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version:Experimental
I use balena etcher to burn "2020-08-20-raspios-buster-armhf- lite.img ", and then use" $bash < (curl - sl https://www.screenly.io/install-ose.sh ）"Installation of the experimental version has failed all the time in run docker compose
[image]

vpetersson, 2020-11-19:
Yeah this is expected. I'm working on this right now. Hope to have something ready in the coming days.

closed 2020-11-19 by vpetersson

wmysummer, 2020-11-19:
> Yeah this is expected. I'm working on this right now. Hope to have something ready in the coming days.

OK，Thanks.

vpetersson, 2020-12-20:
@wmysummer Try upgrading to the latest "developer" release now.