Screenly/Anthias #1426: QA Experimental branch Pi 4

vpetersson, 2020-12-02:
Checklist:

  - [ ] Assets
    - [ ] Web
    - [ ] Images
    - [ ] Video
  - [ ] Balena
  - [ ] Native
    - [ ] Splash Screen

This task shall be completed *after* all other tickets in this milestone.

vpetersson, 2020-12-02:
I've updated [Dockerfile.viewer](https://github.com/Screenly/screenly-ose/blob/experimental/docker/Dockerfile.viewer) to use the QT Base and Webview from #1429. However, it is still failing due to some issue related to hardware acceleration.

# Experiments done
## Default with more logging

```
$ export QT_QPA_EGLFS_INTEGRATION=eglfs_kms
$ ScreenlyWebview
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
qt.qpa.egldeviceintegration: EGL device integration plugin keys: ("eglfs_emu", "eglfs_kms_egldevice", "eglfs_x11")
qt.qpa.egldeviceintegration: EGL device integration plugin keys (sorted): ("eglfs_kms", "eglfs_emu", "eglfs_kms_egldevice", "eglfs_x11")
qt.qpa.egldeviceintegration: Trying to load device EGL integration "eglfs_kms"
qt.qpa.egldeviceintegration: Failed to load EGL device integration "eglfs_kms"
qt.qpa.egldeviceintegration: Trying to load device EGL integration "eglfs_emu"
qt.qpa.egldeviceintegration: Using EGL device integration "eglfs_emu"
Could not initialize egl display
Aborted (core dumped)
```

## Switch EGLFS device

```
$ cat << EOF > /usr/src/app/config.json
{
  "device": "/dev/dri/card1",
  "hwcursor": false,
  "pbuffers": true,
  "outputs": [
    {
      "name": "HDMI1",
      "mode": "1920x1080",
      "primary": false
    },
    {
      "name": "HDMI2",
      "mode": "1920x1080",
      "primary": false,
      "clones": "HDMI1"
    }
  ]
}
EOF
$ export QT_QPA_EGLFS_INTEGRATION=eglfs_kms_egldevice
$ export QT_QPA_EGLFS_KMS_CONFIG=/usr/src/app/config.json
$ ScreenlyWebview
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
qt.qpa.egldeviceintegration: EGL device integration plugin keys: ("eglfs_emu", "eglfs_kms_egldevice", "eglfs_x11")
qt.qpa.egldeviceintegration: EGL device integration plugin keys (sorted): ("eglfs_kms_egldevice", "eglfs_emu", "eglfs_x11")
qt.qpa.egldeviceintegration: Trying to load device EGL integration "eglfs_kms_egldevice"
qt.qpa.eglfs.kms: Loading KMS setup from "/usr/src/app/config.json"
qt.qpa.eglfs.kms: Requested configuration (some settings may be ignored):
        headless: false
        hwcursor: false
        pbuffers: true
        separateScreens: false
        virtualDesktopLayout: 0
        outputs: QMap(("HDMI1", QMap(("mode", QVariant(QString, "1920x1080"))("name", QVariant(QString, "HDMI1"))("primary", QVariant(bool, false))))("HDMI2", QMap(("clones", QVariant(QString, "HDMI1"))("mode", QVariant(QString, "1920x1080"))("name", QVariant(QString, "HDMI2"))("primary", QVariant(bool, false)))))
qt.qpa.eglfs.kms: New DRM/KMS on EGLDevice integration created
qt.qpa.egldeviceintegration: Using EGL device integration "eglfs_kms_egldevice"
qt.qpa.eglfs.kms: platformInit: Opening DRM device
qt.qpa.eglfs.kms: Found 1 EGL devices
qt.qpa.eglfs.kms: Using backend-provided DRM device /dev/dri/card0
Could not open DRM (NV) device
Aborted (core dumped)
```

The permission on said file appears to be fine:
```
$ ls -lah /dev/dri/card0
crw-rw---- 1 root video 226, 0 Dec  1 16:38 /dev/dri/card0
$ whoami
root
```

schens79, 2020-12-16:
Im newbie for RP and Screenly.  I bought RP4 for Christmas and now it seems to be that sad, that i can't get Screenly OSE run on my brand new and first RP? Or is there a chance? Unfortunally i would need a step by step guide and to know if it is possible to run via WiFi. For now I have Pi OS light running and Connection to Internet.

vpetersson, 2020-12-17:
Pi 4 support is under active development. It would be advised that you follow the development on this thread to see when it has been fully closed out before giving it a go.

vpetersson, 2020-12-19:
I'm happy to let everyone know that I've now merged experimental into master and I can confirm that I can run all types of assets on the Pi 4.

Balena support is still to be added.

HendryLeo, 2020-12-21:
Hi, Any timeline to merge this into production?

ealmonte32, 2020-12-21:
> Hi, Any timeline to merge this into production?

Any development version should be thoroughly tested before being considered moved to the production branch.
This version literally just made it to development branch, so there are a lot of testing scenarios and cases that would need to happen prior to this.
If you have a spare SD card and can run the master/dev branch to test if all your assets work fine, you can definitely use it, and any issues you encounter should def give feedback by opening up a detailed issue of it.

ealmonte32, 2020-12-21:
> I'm happy to let everyone know that I've now merged experimental into master and I can confirm that I can run all types of assets on the Pi 4.
> 
> Balena support is still to be added.

I just got on the Pi4, tried to run regular install script, select dev/master branch, the pulling of images were all "done", then got this error at the `up` attempt:

```
+ sudo -E docker-compose -f /home/pi/screenly/docker-compose.yml -f /home/pi/screenly/docker-compose.override.yml up -d
Creating network "screenly_default" with the default driver
Creating volume "screenly_screenly-data" with default driver
Creating volume "screenly_resin-data" with default driver
Creating screenly_srly-ose-server_1 ... 
Creating screenly_redis_1           ... error

Creating screenly_srly-ose-server_1 ... error
air interfaces: operation not supported

ERROR: for screenly_srly-ose-server_1  Cannot start service srly-ose-server: failed to create endpoint screenly_srly-ose-server_1 on network screenly_default: failed to add the host (vetha634d15) <=> sandbox (veth8520d5d) pair interfaces: operation not supported

ERROR: for redis  Cannot start service redis: failed to create endpoint screenly_redis_1 on network screenly_default: failed to add the host (veth0194544) <=> sandbox (vethad9ba80) pair interfaces: operation not supported

ERROR: for srly-ose-server  Cannot start service srly-ose-server: failed to create endpoint screenly_srly-ose-server_1 on network screenly_default: failed to add the host (vetha634d15) <=> sandbox (veth8520d5d) pair interfaces: operation not supported
ERROR: Encountered errors while bringing up the project.
```

ealmonte32, 2020-12-21:
Regarding Splashscreen:

After lots of searching and much troubleshooting, a simple reinstall of the plymouth package fixes the splashscreen issue.
Meaning, right after the installation finishes, upon reboot, a `sudo apt install plymouth --reinstall` was the only 100% reproducible way I was able to get it back to work without changing anything else. Other methods were hit and miss.. so I was wondering, which of these processes that are triggered after the package gets installed is the one that makes it work... and I know it comes down to the initramfs because systemd service doesnt change from initial install and change of theme, and man-db is obviously manual related so that's out..

Something with `update-initramfs` is probably what is done for the "Processing triggers for initramfs-tools.." , but during the troubleshooting and running `update-initramfs -u -v`, it said: "Nothing to do, exiting." So I have a feeling it doesn't do anything without putting some more flags/arguments into the command..
 
I also looked at the hooks that are ran from `/usr/share/initramfs-tools/hooks/plymouth` but I cant pinpoint what part is the one correcting the issue..
I think I even tried to manually run this script to see if it was the trigger that fixes it, but I forgot if this was also a hit or miss result..

---

Reference, plymouth installation triggers:
```
update-initramfs: deferring update (trigger activated)
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for systemd (241-7~deb10u4+rpi1) ...
Processing triggers for initramfs-tools (0.133+deb10u1) ...
```

vpetersson, 2020-12-22:
Good findings @ealmonte32!

vpetersson, 2020-12-30:
I've added [this page](https://github.com/Screenly/screenly-ose/blob/master/docs/roadmap.md#known-issues-in-master-branch) to track outstanding issues.

JasonNVSD, 2021-01-14:
> I'm happy to let everyone know that I've now merged experimental into master and I can confirm that I can run all types of assets on the Pi 4.

I'd like to test this on a Pi 4.  I booted the current Pi OS Lite, then ran install-ose.sh.  I answered 2 for master, yes to manage network, no to system update.  Are these the correct choices?

It seemed to install correctly, but when I boot it, I don't see the Screenly welcome screen.  Alt+F1 brought up the console.  I got the IP address with ifconfig.  I found it responded to a ping, but no http response.  What should I try next?

vpetersson, 2021-01-14:
> Alt+F1 brought up the console. 

This means the installation failed. You shouldn't be able to do this.

Here's what I would recommend that you do. 

* Enable SSH on the Pi if you haven't already.
* Re-run `/home/pi/screenly/bin/run_upgrade.sh`
* If it still doesn't work, then post the output of the following command: `cd /home/pi/screenly && docker-compose logs`

JasonNVSD, 2021-01-14:
Re-running didn't fix it.  Log is attached.

[log.txt](https://github.com/Screenly/screenly-ose/files/5816323/log.txt)

vpetersson, 2021-01-14:
When did you run this? Curious as that bug was resolved earlier today.

JasonNVSD, 2021-01-14:
> When did you run this? Curious as that bug was resolved earlier today.

About 30 minutes ago.

ealmonte32, 2021-01-14:
@JasonNVSD did you re-run through `bash <(curl...` command or just re-ran the actual file install.sh that was on the `screenly/bin/` directory? maybe the screenly repo directory on the Pi is not up to date with the upstream? I would say to remove the local `/home/pi/screenly/` directory just so that when you run the install-ose.sh file the cloned repo is up to date..

JasonNVSD, 2021-01-14:
> @JasonNVSD did you re-run through `bash <(curl...` command or just re-ran the actual file install.sh that was on the `screenly/bin/` directory? maybe the screenly repo directory on the Pi is not up [...]

I deleted the directory and ran the install command again.  It's working now.  Thank you.

If I'm doing this again from scratch, what steps exactly should I be doing after installing Pi OS Lite?

ealmonte32, 2021-01-14:
> I deleted the directory and ran the install command again. It's working now. Thank you.

No prob, I'm glad it was that simple.

___

> If I'm doing this again from scratch, what steps exactly should I be doing after installing Pi OS Lite?

Nothing really, if you are doing it from scratch then the Raspbian OS will not include the screenly folder with the contents of the repo, that happens during the install when the ansible command using git calls on to the github repo and clones it to your Pi, thus any time you do this from scratch it will always be the up-to-date files that will be installed, get it?

JasonNVSD, 2021-01-14:
> Nothing really, if you are doing it from scratch then the Raspbian OS will not include the screenly folder with the contents of the repo, that happens during the install when the ansible command [...]

Yeah, I don't have a linux background, but I figured as much.  I tried from scratch (starting with Pi Imager) twice yesterday, and twice today.  I remember in an old thread here someone talking about running "apt update" and such to get things to work--is that needed or helpful?

ealmonte32, 2021-01-14:
> I remember in an old thread here someone talking about running "apt update" and such to get things to work. Is that needed or helpful?

Well, the install-ose.sh script does this as part of its steps so you dont need to do it, `apt update` simply updates the list of package information (versions and dependencies, etc) from the sources. Your issue had nothing to do with it.

JasonNVSD, 2021-01-14:
I have two issues to report.  During boot there's twice that it echos the following.  I don't know what it means or if it matters.

[FAILED] Failed to start Show Plymouth Boot Screen.

And more importantly, clicking System Info in the web console doesn't work and sometimes causes it to stop responding.

ealmonte32, 2021-01-14:
> [FAILED] Failed to start Show Plymouth Boot Screen

Part of my troubleshooting on Pi4 is trying to solve that on a fresh install so this one is a known issue..

> And more importantly, clicking System Info in the web console doesn't work and sometimes causes it to stop responding.

Not sure about this one, System info on the web as far as I know does not show logs like the production version did because the logs are inside a container, there is discussion on maybe having to create a small container service that serves only this purpose of showing logs. Not sure what you mean by causes it to stop responding since the system-info only retrieves info/data, it doesn't push or write data thus not being able to cause anything..

vpetersson, 2021-01-15:
The 'System Info' issue is a known one. I'm aware of it but haven't been able to fully resolve it yet. It has something to do with the new environment, but I'm not getting anything useful in the logs and need to keep probing.

ealmonte32, 2021-01-15:
> The 'System Info' issue is a known one. I'm aware of it but haven't been able to fully resolve it yet. It has something to do with the new environment, but I'm not getting anything useful in the logs and need to keep probing.

Unless something within the last few weeks changed with server side, I haven't experienced this at all on Pi4 once it has been up and running, (only when troubleshooting and reinstalling and restarting containers live, which I thought was a docker issue and only restarting the Pi fixes it), maybe I am getting confused what the user means by "causes it to stop responding", how can loading the system-info page which just retrieves data cause the Pi to become unresponsive? Unless what is happening is that the server container is becoming unresponsive and thus making it seem like the Pi is unresponsive.. is this more or less the issue? dealing with the web server container?

vpetersson, 2021-01-15:
Yeah so I'm fairly sure this is related to the CEC changes I brought in. At least that is the current hypothesis. In short, I changed the way we read display status to libcec instead, which should be more reliable. However, for whatever reason it times out on the systems page, but works fine when running it from the command line using `diagnostics.py`. So the webserver will basically timeout.

JasonNVSD, 2021-01-15:
> maybe I am getting confused what the user means by "causes it to stop responding", how can loading the system-info page which just retrieves data cause the Pi to become unresponsive?
I just meant the web interface stopped responding.  It doesn't happen consistently.  I'll just stop clicking on it. :p

JasonNVSD, 2021-01-15:
With the Sprint 10.1 disk image on a Pi 3, I configured them with a static IP on eth0.  I found out that it required using an older method of editing /etc/network/interfaces.  This has been working great.

When I tried this on the development build it had problems.  So I tried the newer method of editing /etc/dhcpcd.conf [per the Pi documentation](https://www.raspberrypi.org/documentation/configuration/tcpip/README.md), which also had problems.  I tried a lot of different things I googled without success.  I didn't keep track of them all.

I started from scratch and chose "no" to network management.  Then I configured it per the Pi documentation.  This is working with a couple of weird caveats.

- The splash screen shows the old DHCP address and not the static address.  This persists after multiple reboots and retries.
- The assets don't display after booting.  If I toggle an asset off and back on, then it displays.  Alternatively, if I disable the splash screen, the assets will display after a boot.

I realize this isn't really supported, but I'm probably not the only one who will want to do it.

vpetersson, 2021-01-16:
> The splash screen shows the old DHCP address and not the actual address. This persists after multiple reboots and retries.

Yeah this is expected. I put in a bit of a hackish workaround for now. Basically, because we can't read the host IP from within Docker, I pass it on as part of the installation process. This needs to be reworked in the future, but not exactly sure yet.

> The assets don't display after booting. If I toggle an asset off and back on, then it displays. Alternatively, if I disable the splash screen, the assets will display after a boot.

That's odd. Can you dig out some logs for this?

JasonNVSD, 2021-01-18:
> That's odd. Can you dig out some logs for this?

It's up on a sign now (with the splash screen disabled).  If I pull it down, what logs and how?

vpetersson, 2021-01-19:
This should do:
```
$ cd /home/pi/screenly
$ docker-compose logs
[...]
$ sudo journalctl --no-pager
```

ealmonte32, 2021-01-22:
@vpetersson 
with regards to splash screen, do you want to add the following command at near end of install.sh as temporary workaround in order to fix the issue for now?

`sudo apt install plymouth --reinstall`

this solves the problem for now and I'm just thinking to do it for now because it's been really hard to pinpoint why it exactly happens.. and as you know it has no side effects whatsoever since its a reinstall.

I also wanted to opine about the removal of experimental branch, for example, master was working well before CEC addition, do you think this experimental branch should be brought back in order to mess with such changes/additions/etc so that we have a sort of "fully working" master/dev branch?

ealmonte32, 2021-01-22:
> Yeah so I'm fairly sure this is related to the CEC changes I brought in. At least that is the current hypothesis. In short, I changed the way we read display status to libcec instead, which should [...]

I can confirm that commenting out the `get_display_power()` functions from `server.py` allows for system-info to load again..

ealmonte32, 2021-01-22:
> * The splash screen shows the old DHCP address and not the actual address.  This persists after multiple reboots and retries.

one of the hackish ways I've been testing to easily get this between host and container has been to run the MY_IP variable command but send the contents to a file as well that only contains this info and then copy that file to the container and from container pull it into the variable.. like this:

from the pi host:
```
pi@screenlypi4:~ $ MY_IP=$(ip -4 route get 8.8.8.8 | awk {'print $7'} | tr -d '\n')
pi@screenlypi4:~ $ echo $MY_IP > myipfile
pi@screenlypi4:~ $ cat myipfile 
192.168.1.15

pi@screenlypi4:~ $ docker cp myipfile screenly_srly-ose-server_1:/usr/src/app/
```

get into the container now and then edit the start_server.sh file:
```
pi@screenlypi4:~ $ docker exec -it screenly_srly-ose-server_1 bash

just to test the file contents:
root@7999d5940ab8:/usr/src/app# cat myipfile 
192.168.1.15

if you havent installed nano, you need to install it with: apt update && apt install nano
root@7999d5940ab8:/usr/src/app# nano bin/start_server.sh 

add this line somewhere at the top so it looks like this:
#!/bin/bash
export MY_IP=$(cat /usr/src/app/myipfile)
```
save it by pressing ctrl+x then Y.. then exit

`root@7999d5940ab8:/usr/src/app# exit`

on the host, restart the server and viewer container or reboot..
`pi@screenlypi4:~ $ docker container restart screenly_srly-ose-server_1 screenly_srly-ose-viewer_1 `
or `sudo reboot`

ealmonte32, 2021-01-22:
@vpetersson the above can also work for the mac address as well since system-info needs it, and we can use pip module `getmac==0.8.2`, then run `getmac > mymacaddress` and then copy that into the server container and export it as a env `MY_MAC` and then call it from `lib/utils.py` as an os.getenv ... i know, it's ridiculous but unless there is a package that does all this automatically from inside the docker to host with correct interface, this is one of the ways I've customized the environment which just needs some automated celery task or something that does this upon every reboot for both the ip address file and mac address file..

vpetersson, 2021-01-22:
> @vpetersson
> with regards to splash screen, do you want to add the following command at near end of install.sh as temporary workaround in order to fix the issue for now?
>
> `sudo apt install plymouth [...]

Yeah let's do this.

> I also wanted to opine about the removal of experimental branch, for example, master was working well before CEC addition, do you think this experimental branch should be brought back in order to mess with such changes/additions/etc so that we have a sort of "fully working" master/dev branch?

I disagree. It's too much overhead with two active development branches. We'll inevitably end up in 'merge conflict' territory again. The developer branch is intended to be 'moving fast' and sometimes break. Whilst it is subpar that we can't show the System Info page, it doesn't prevent usage. 

> I can confirm that commenting out the `get_display_power()` functions from `server.py` allows for system-info to load again..

Ok I've pushed a fix that hopefully works around the issue for it. In the process of testing it.

> one of the hackish ways I've been testing to easily get this between host and container has been to run the MY_IP variable command but send the contents to a file as well that only contains this [...]

Sure. There are plenty of ways to pass on information from the host, but what they all have in common is that they require a service to run on the host. What you're suggesting is similar to what's being done in the installation script. If we had a service running on the host directly, we could relatively easy do this. The problem however is that it doesn't sit well with me as it breaks the concept of everything running inside Docker.

Now, we may inevitably need cross this road anyways with some kind of supervisor (similar to [balena supervisor](https://github.com/balena-io/balena-supervisor)), that is essentially an API server for communicating to the host. We will need this to perform things like:

* System reboots
* System halts
* Retrieve system logs

P.s. I believe an easier way to do the same thing is to just run these commands:

```
export MY_IP=$(ip -4 route get 8.8.8.8 | awk {'print $7'} | tr -d '\n')
TOTAL_MEMORY_KB=$(grep MemTotal /proc/meminfo | awk {'print $2'})
export VIEWER_MEMORY_LIMIT_KB=$(echo "$TOTAL_MEMORY_KB" \* 0.7 | bc)

sudo -E docker-compose \
    -f /home/pi/screenly/docker-compose.yml \
    -f /home/pi/screenly/docker-compose.override.yml \
    up -d
```

> @vpetersson the above can also work for the mac address as well since system-info needs it, and we can use pip module `getmac==0.8.2`, then run `getmac > mymacaddress` and then copy that into the [...]

Right. So getting the MAC is easy. The problem is to pass it on. It is essentially the same issue as getting the IP address: we can't read the state of the interface from within the container. The remedy is thus the same: either passing it on as an environment variable, or some sort of supervisor.

The issue with passing it on as an environment variable as you correctly pointed out is that it is static, so if you for instance get a new IP from the DHCP server, or switch from WiFi to Ethernet, neither the IP or MAC will be correct.

Using Celery for this is an interesting idea. It might work actually, but we would need to run it in 'Host mode', which basically gives Celery free reign to do anything on the system.

magno23, 2021-02-05:
is internet connection really necessary now? i have been using my rpi3b with screenly production@618864d , configured to use ntp time from my server and so far with 0 issues, i'm trying to add my rpi4 with the latest dev/master build and it only works if i enable internet connection to the network, otherwise i get 'internal server error' on the browser

vpetersson, 2021-02-05:
> is internet connection really necessary now? i have been using my rpi3b with screenly production@618864d , configured to use ntp time from my server and so far with 0 issues, i'm trying to add my [...]

We haven't really tested it for offline mode to be honest, as that is a fairly rare use case.

magno23, 2021-02-05:
> We haven't really tested it for offline mode to be honest, as that is a fairly rare use case.

without docker and balena what other connections are made by the system to the internet? afaik updates are disabled via browser, would be something to do with nginx?

magno23, 2021-02-08:
> 
> 
> > without docker and balena what other connections are made by the system to the internet? afaik updates are disabled via browser, would be something to do with nginx?
> 
> So if I understand correctly you first allow it internet so that it can be installed and all that, then you want to disallow it to access external servers and that your browser says "internal server error"? when does it say this, what asset are you adding to it? I would assume anyone with "offline" screenly would need to have all assets be locally uploaded or running internal servers that are accessible from the RPi..

What i did on my old RPi3 was, use the available image at the time, configured the ntp settings to get the time from my server(that is on the same network as the RPi) and added videos and images locally as well as a url point to a webpage located in my server.
On my new RPi4 i used raspbian buster lite, connected it to the internet and followed the instructions to manually install screenly(developer branch and no network manager), disconnected it from the internet and connected it to the same network as my old RPi, connected to it via ssh and edited the ntp settings, tried to access it via browser on my pc and i get "internal server error", if i connect the RPi4 to a network with access to the internet it works OK.
I don't know if my explanation makes sense.

ealmonte32, 2021-02-08:
Can you check the logs to see any errors, and also the status of containers to check if the web server and all others loaded properly?

`docker container ls` to see the status of containers
`journalctl` to see logs

magno23, 2021-02-09:
here is the output of those commands
journalctl -> https://pastebin.com/hrYgYfuE
docker container ls -> https://pastebin.com/j3kfEynv

ealmonte32, 2021-02-10:
> here is the output of those commands
> journalctl -> https://pastebin.com/hrYgYfuE
> docker container ls -> https://pastebin.com/j3kfEynv

well, the docker containers are up so it is not an issue with screenly server running..

this seems to be a networking issue where your pi containers are not getting the proper IP address (via bridge mode) thus not able to route any traffic properly..

example:
```
Feb 05 18:08:00 raspberrypi dhcpcd[366]: wlan0: connected to Access Point `DEVICEACCESS'
Feb 05 18:08:01 raspberrypi dhcpcd[366]: wlan0: rebinding lease of 10.213.8.4
Feb 05 18:08:01 raspberrypi dhcpcd[366]: wlan0: probing address 10.213.8.4/24
```
That seems to be a normally obtained IP to the pi from dhcp, even if there is no access to an outside internet, but then docker is giving the containers self-assigned link-local addresses that should be in the 172.x.x.x subnet:
```
Feb 05 18:08:29 raspberrypi dhcpcd[549]: veth2137579: using IPv4LL address 169.254.104.10
Feb 05 18:08:29 raspberrypi dhcpcd[549]: veth2137579: adding route to 169.254.0.0/16
```
hmmm... I am trying to read more about how docker handles the different networks because this is obviously something dealing with that, like for your specific offline situation, there must be a setting that needs to be set differently so that the containers can still communicate even when there is no access to internet (because this isn't a requirement for containers to work as you can see they are up and running), but the network still needs to have some form of setup which allows routing within that internal network..

magno23, 2021-05-06:
i have not been able to play video assets (.mp4), is there any way to know what I'm doing wrong since the first comment says the videos should play OK.

vpetersson, 2021-05-06:
@magno23 Not all videos are supported. The mere file extension is not sufficient to determine this. The first step would be to run `ffprobe` on the file to ensure it's encoded as h264 using `ffmpeg`.

magno23, 2021-05-06:
@vpetersson i believe it is encoded as h264
here is the output of `ffprobe video.mp4`
```
 Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.35.101
  Duration: 00:01:27.42, start: 0.000000, bitrate: 5692 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], 5690 kb/s, 24 fps, 24 tbr, 12288 tbn, 48 tbc (default)
    Metadata:
      handler_name    : ISO Media file produced by Google Inc. 
```

vpetersson, 2021-05-06:
@magno23 Try with [this video](https://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_1080p_h264.mov) to see if you are able to reproduce it as that should be a known-good one.

magno23, 2021-05-07:
@vpetersson that video worked, what should I do to my videos for them to work also? what codec or encoder should I use?

vpetersson, 2021-05-07:
If that video work, you'll need to reencode your videos. I don't recall the exact settings, but there should be plenty of blog posts on this across the interwebs.

closed 2022-01-09 by stale[bot]