Screenly/Anthias #1427: QA Experimental branch Pi 3

vpetersson, 2020-12-02:
Checklist:

  - [ ] Assets
    - [ ] Web
    - [ ] Images
    - [ ] Video
  - [ ] Balena
  - [ ] Native
    - [ ] Splash Screen

This task shall be completed *after* all other tickets in this milestone.

vpetersson, 2020-12-02:
This is blocked by https://github.com/Screenly/screenly-ose/issues/1429 as there is no available QT Base build.

ealmonte32, 2020-12-03:
> This task shall be completed _after_ all other tickets in this milestone.

I am installing experimental on pi3 now to test these, but I noticed this line at end of this and other post for pi4, can you verify what you mean by this?

ealmonte32, 2020-12-03:
Feedback, just installed current experimental on pi3:

- [x] Assets
  - [x]  Web
  - [x]  Images
  - [x]  Video
- [ ]  Balena
- [ ] Native
  - [ ]  Splash Screen

Notes:
- Web related
  - `https://html5test.com/` set as an asset, gives browser very low score, 256, video 0/33, audio 0/30
  - google maps even in 2d canvas mode does not work
  - grafana prometheus dashboard doesnt load at all, just black screen
- splash screen does not work right after install, very annoying as the debug is hard since the logs dont tell you exactly why the plymouth-start.service fails.. i cannot figure out why (even after reading all the initramfs-hooks and install triggers) why just running a reinstall of plymouth package makes it work.. ughhh.. maybe we need to put the files in their respective directories first, then run ansible task for installing plymouth.. I will try this and see how it goes.

vpetersson, 2020-12-03:
@ealmonte32 ah, yeah so this is an old cached image. It's not using QT 5.15.2, so let's ignore that QC run. If you try to build the image from the branch, it will fail:

```
$ docker build \
    --build-arg PI_VERSION=pi3 \
    -t screenly/srly-ose-viewer:experimental-pi3 \
    -f docker/Dockerfile.viewer .
```

ealmonte32, 2020-12-03:
So I tried to follow along with the notes you left for building the webview, but i ended up in a github repo that was in french 🤷🏻‍♂️ 

Can you tell me if I am completely in the wrong path here, I've only compiled a few packages from source, so I am a bit lost doing this QT webview now..

I dont mind leaving a pi3 running to build, and a pi zero w as well, since you said it takes 3hrs to build in an i7 machine..

Ok, so, in layman's terms, since webview src folder is in the experimental branch only, I get it by typing this:
`git clone -b experimental https://github.com/Screenly/screenly-ose.git`

then, once I am in that `webview` directory, according to your guide, I type this:
`$ docker build -t qt-builder .`

then, once that finishes, I type:
```
docker run --rm -t \
    -v ~/tmp/qt-src:/src \
    -v ~/tmp/qt-build:/build \
    qt-builder
```

Am I doing this on the Pi or my actual computer?
Also, once this finishes building, what am I doing with the files in `~/tmp/qt-build/` ?
Are these steps how am I building a new webview with QT 5.15.2 ?
Once it finishes, how exactly is the new qt webview package/file/binary/container? going to replace this old Stretch+QT 5.9 build? is this process running dockerfile and qt_build from scratch to get a container that will have packages with QT 5.12.2?
🤔

vpetersson, 2020-12-03:
> So I tried to follow along with the notes you left for building the webview, but i ended up in a github repo that was in french 🤷🏻‍♂️

:)

> Am I doing this on the Pi or my actual computer?

You need an actual computer for this, and a beefy one at that. I've only tested this on macOS and Linux. Note that you need the experimental features enabled as per the instructions in the ticket.

> Are these steps how am I building a new webview with QT 5.15.2 ?

Right, so the steps in the README are for how to build QT 5.12.2.

> Once it finishes, how exactly is the new qt webview package/file/binary/container? going to replace this old Stretch+QT 5.9 build? is this process running dockerfile and qt_build from scratch to get a container that will have packages with QT 5.12.2?

For Pi 4, you can use the builds that I have uploaded already. They will be retrieved automatically. I'm not 100% sure they are properly compiled as I'm still battling with some EGLS issues.

The first step for getting the Pi 3 working can be found [here](https://github.com/Screenly/screenly-ose/issues/1429). Basically, you need to get the `./configure` step in QT Base to show `EGLFS Rasberry Pi .................... yes`. Otherwise hardware acceleration will not work.

> I just saw the Pi3 stop with an error after a while of building with typing docker build -t qt-builder .
This is the error:

Yeah don't even attempt to build this on the Pi 3... :)

vpetersson, 2020-12-04:
Success! 

```
Building on: linux-g++ (x86_64, CPU features: mmx sse sse2)
Building for: devices/linux-rasp-pi3-g++ (arm, CPU features: neon)
Target compiler: gcc 7.4.1
[...]
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. yes
  EGLFS details:
    EGLFS OpenWFD ........................ no
    EGLFS i.Mx6 .......................... no
    EGLFS i.Mx6 Wayland .................. no
    EGLFS RCAR ........................... no
    EGLFS EGLDevice ...................... no
    EGLFS GBM ............................ no
    EGLFS VSP2 ........................... no
    EGLFS Mali ........................... no
    EGLFS Raspberry Pi ................... yes
    EGLFS X11 ............................ no
  LinuxFB ................................ yes
  VNC .................................... yes
[...]
```

Had to apply these changes:
* https://github.com/Screenly/screenly-ose/commit/54492afe5f84f5d58433accfbdef866523b18409
* https://github.com/Screenly/screenly-ose/commit/bd67cdcefbf62c093a725f28545b3ebcb30302b1

vpetersson, 2020-12-19:
I have now merged experimental into master and I'm happy to confirm that it is now working. There are a lot more things to be tested. There are a number of outstanding issues, but I've been mainly focused on getting everything working int he first place.

ealmonte32, 2020-12-21:
> I have now merged experimental into master and I'm happy to confirm that it is now working. There are a lot more things to be tested. There are a number of outstanding issues, but I've been mainly focused on getting everything working int he first place.

Awesome.
Will be testing and troubleshooting on Pi3 first.

ealmonte32, 2020-12-21:
Getting [this](https://github.com/docker/docker-py/issues/1502) error while trying to install on a Pi3 that already had screenly-ose master branch installed:

```
+ sudo -E docker-compose -f /home/pi/screenly/docker-compose.yml -f /home/pi/screenly/docker-compose.override.yml up -d
Traceback (most recent call last):
  File "/usr/local/bin/docker-compose", line 5, in <module>
    from compose.cli.main import main
  File "/usr/local/lib/python2.7/dist-packages/compose/cli/main.py", line 18, in <module>
    import docker.errors
...
...
...
  File "/usr/local/lib/python2.7/dist-packages/docker/transport/__init__.py", line 3, in <module>
    from .ssladapter import SSLHTTPAdapter
  File "/usr/local/lib/python2.7/dist-packages/docker/transport/ssladapter.py", line 23, in <module>
    from backports.ssl_match_hostname import match_hostname
ImportError: No module named ssl_match_hostname
```

So, according to [this](https://github.com/docker/docker-py/issues/1502#issuecomment-506544849) comment I checked to see if option 2 was even something to consider, but then I saw results of a reverse depends:
```
pi@rpi3blk:~ $ apt-cache rdepends python-configparser 
python-configparser
Reverse Depends:
  elastalert
  sagemath
  python-yapsy
  python-nipype
  python-metaconfig
  python-future
  python-flake8
  python-entrypoints
  python-custodia
  pylint
  nagios2mantis
  ibus-braille
  hinge
  eyes17
```

So I as well decided to go with option 1 (installed `python-backports.ssl-match-hostname` with apt), uninstalling from pip made no difference since it gets installed again as a dependency for `docker-compose`, but as soon as that one backports module error went away, the other backports module gave the same error:
```
ImportError: No module named shutil_get_terminal_size
```
..so I installed it from debian source: `python-backports-shutil-get-terminal-size`

..then it worked..

@vpetersson , what do you think should be the next step here.. either install the backports using apt, or some other users solved it by copying the contents of the pip installed libraries into the non local directory.. which is kinda ugly to do..

vpetersson, 2020-12-21:
@ealmonte32 What Raspbian version is this on? I haven't been able to reproduce this myself.

ealmonte32, 2020-12-23:
> @ealmonte32 What Raspbian version is this on? I haven't been able to reproduce this myself.

Buster, but I think it was something I installed during troubleshooting other packages that probably installed it via apt instead of pip, which explains the issue, so we can disregard.

vpetersson, 2020-12-23:
Thanks! Yeah, I did a fresh install on Buster Lite two days ago and it worked well for me.

vpetersson, 2020-12-30:
I've added [this page](https://github.com/Screenly/screenly-ose/blob/master/docs/roadmap.md#known-issues-in-master-branch) to track outstanding issues.

ealmonte32, 2021-01-01:
With regards to the logs and systemd, I was able to look at logs by running this cmd:
`docker container logs --tail 20 screenly_srly-ose-viewer_1`

because this command doesnt work:
`journalctl -b CONTAINER_NAME=screenly_srly-ose-viewer_1 -n 20`

vpetersson, 2021-01-04:
Right, but that will only work *outside* the containers. Basically, we need some way to either connect to Dbus. It [appears to be possible](https://trstringer.com/python-systemd-dbus/), but I haven't explored it. Most likely we need a new service that will only work as a proxy for dbus to limit the attack vector. This service would just have a simple API for things like getting logs and reboot the system.

closed 2022-01-09 by stale[bot]