Screenly/Anthias #1441: Balena issues with most recent build

singh87, 2020-12-04:
#### Overview of the Issue

So this all started a week back when I updated my screenly-ose because my sd card had corrupted. I purchased a high quality sd card this time around and proceeded with flashing balenaOS and installing screenly. I was never able to successfully deploy to balena because of [#1431](https://github.com/Screenly/screenly-ose/issues/1431). This seems to have been fixed with the most recent commit.

The issue I'm having now is that redis keeps exiting/crashing in balena and I have no idea how to fix it.
[image]

#### Steps

1. Flash latest BalenaOS to sd card
2. Balena push most recent commit from master branch
3. Redis keeps restarting at this point
4. Public url for balena stops working after recovering assets from backup and rebooting the device

#### Environment

* Raspberry Pi Hardware Version: Pi Zero W
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: lastest Dec 3, 2020 commit

vpetersson, 2020-12-04:
Strange. Looks like a communication error. Make sure the redis container is working properly.

vpetersson, 2020-12-04:
Found and fixed the error in master.

closed 2020-12-04 by vpetersson

singh87, 2020-12-04:
Ok so I'm still getting this issue. It seemed to be fixed with the newest commit but when I upload my backup assets to screenly, I get the error message "The operation failed. Please reload the page and try again.". Reloading the page didn't help so I restarted the device and get the redis error again every 32 seconds.

singh87, 2020-12-05:
@vpetersson Could you please reopen this?

vpetersson, 2020-12-05:
Can't really spend cycles on this. All my focus is on getting the experimental branch merged into master. Will look into this if it is still an issue after that.

singh87, 2020-12-10:
I completely understand. I actually ended up ordering a raspberry pi 3b+ and balena/screenly work fine on there. I'm guessing the pi-zero just isn't powerful enough to run the latest build anymore.