Screenly/Anthias #1455: Issue on balenaCloud with Pi 4 / 400

shawaj, 2020-12-20:
#### Overview of the Issue

Permissions errors in viewer.py file causing the viewer container to keep craching.

#### Reproduction Steps

Steps to reproduce this issue:

* Deploy on balenaCloud with a Pi4 / 400
* Add assets using OSE backend
* Activate assets
* Screen goes white and then every few minutes flashes black and then back white again (at same time as viewer container crashes in logs

#### Environment

* Raspberry Pi Hardware Version: Pi 4 or Pi 400
* Raspberry Pi Network Setup: built in WiFi or Ethernet
* Screenly OSE Version: latest from github

#### Log files

```
20.12.20 15:54:35 (+0000) Service exited 'srly-ose-viewer sha256:725905924498c0187f1346c324a0615f45efd6e86ab80f8dcc245956a3ddb90d'
20.12.20 15:55:36 (+0000) Restarting service 'srly-ose-viewer sha256:725905924498c0187f1346c324a0615f45efd6e86ab80f8dcc245956a3ddb90d'
20.12.20 15:55:36 (+0000)  srly-ose-viewer  chown: cannot access '/data/.screenly/latest_screenly_sha': No such file or directory
20.12.20 15:55:36 (+0000)  srly-ose-viewer  chown: cannot access '/etc/sudo.conf': No such file or directory
20.12.20 15:55:37 (+0000)  srly-ose-viewer  Loading browser...
20.12.20 15:55:39 (+0000)  srly-ose-viewer  Generating asset-list...
20.12.20 15:55:39 (+0000)  srly-ose-nginx  172.17.0.4 - - [20/Dec/2020:15:55:39 +0000] "GET / HTTP/1.1" 200 27155 "-" "python-requests/2.24.0" "-"
20.12.20 15:55:39 (+0000)  srly-ose-viewer  Current url is http://srly-ose-nginx:80/static/img/loading.png
20.12.20 15:55:39 (+0000)  srly-ose-viewer  Viewer crashed.
20.12.20 15:55:39 (+0000)  srly-ose-viewer  Traceback (most recent call last):
20.12.20 15:55:39 (+0000)  srly-ose-viewer    File "viewer.py", line 529, in <module>
20.12.20 15:55:39 (+0000)  srly-ose-viewer      main()
20.12.20 15:55:39 (+0000)  srly-ose-viewer    File "viewer.py", line 524, in main
20.12.20 15:55:39 (+0000)  srly-ose-viewer      asset_loop(scheduler)
20.12.20 15:55:39 (+0000)  srly-ose-viewer    File "viewer.py", line 381, in asset_loop
20.12.20 15:55:39 (+0000)  srly-ose-viewer      check_update()
20.12.20 15:55:39 (+0000)  srly-ose-viewer    File "viewer.py", line 325, in check_update
20.12.20 15:55:39 (+0000)  srly-ose-viewer      with open(device_id_file, 'w') as f:
20.12.20 15:55:39 (+0000)  srly-ose-viewer  IOError: [Errno 13] Permission denied: '/data/.screenly/device_id'
20.12.20 15:55:39 (+0000)  srly-ose-viewer  Traceback (most recent call last):
20.12.20 15:55:39 (+0000)  srly-ose-viewer    File "viewer.py", line 529, in <module>
20.12.20 15:55:39 (+0000)  srly-ose-viewer      main()
20.12.20 15:55:39 (+0000)  srly-ose-viewer    File "viewer.py", line 524, in main
20.12.20 15:55:39 (+0000)  srly-ose-viewer      asset_loop(scheduler)
20.12.20 15:55:39 (+0000)  srly-ose-viewer    File "viewer.py", line 381, in asset_loop
20.12.20 15:55:39 (+0000)  srly-ose-viewer      check_update()
20.12.20 15:55:39 (+0000)  srly-ose-viewer    File "viewer.py", line 325, in check_update
20.12.20 15:55:39 (+0000)  srly-ose-viewer      with open(device_id_file, 'w') as f:
20.12.20 15:55:39 (+0000)  srly-ose-viewer  IOError: [Errno 13] Permission denied: '/data/.screenly/device_id'
20.12.20 15:55:40 (+0000)  srly-ose-viewer  ./bin/start_viewer.sh: line 47: kill: (38) - No such process
20.12.20 15:55:41 (+0000) Service exited 'srly-ose-viewer sha256:725905924498c0187f1346c324a0615f45efd6e86ab80f8dcc245956a3ddb90d'
```

Let me know if you need any more details.

vpetersson, 2020-12-21:
Thanks @shawaj. This is actually two separate bugs:

* `chown: cannot access '/data/.screenly/latest_screenly_sha': No such file or directory` - This itself is not a problem per se, but the fact that the file doesn't exist means that other things will break. This should however be a Balena specific issue. The solution is likely to move this metadata to somewhere else now that we run  inside a container.
* `IOError: [Errno 13] Permission denied: '/data/.screenly/device_id'` this is something that I haven't encountered before but will need to address. I know why it's firing. Just need to work around it.

vpetersson, 2020-12-21:
@shawaj This should have been fixed in master now.

closed 2020-12-21 by vpetersson

shawaj, 2020-12-22:
@vpetersson now I am getting some new errors:

```
22.12.20 02:22:17 (+0000)  srly-ose-viewer  Loading browser...
22.12.20 02:22:19 (+0000)  srly-ose-viewer  Generating asset-list...
22.12.20 02:22:19 (+0000)  srly-ose-viewer  Current url is http://srly-ose-nginx:80/splash-page
22.12.20 02:23:19 (+0000)  srly-ose-viewer  Current url is http://srly-ose-nginx:80/static/img/loading.png
22.12.20 02:23:19 (+0000)  srly-ose-viewer  Viewer crashed.
22.12.20 02:23:19 (+0000)  srly-ose-viewer  Traceback (most recent call last):
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "viewer.py", line 465, in <module>
22.12.20 02:23:19 (+0000)  srly-ose-viewer      main()
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "viewer.py", line 460, in main
22.12.20 02:23:19 (+0000)  srly-ose-viewer      asset_loop(scheduler)
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "viewer.py", line 317, in asset_loop
22.12.20 02:23:19 (+0000)  srly-ose-viewer      is_up_to_date()
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "/usr/src/app/lib/github.py", line 73, in is_up_to_date
22.12.20 02:23:19 (+0000)  srly-ose-viewer      latest_sha, retrieved_update = fetch_remote_hash()
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "/usr/src/app/lib/github.py", line 58, in fetch_remote_hash
22.12.20 02:23:19 (+0000)  srly-ose-viewer      latest_sha = resp.json()['object']['sha']
22.12.20 02:23:19 (+0000)  srly-ose-viewer  TypeError: list indices must be integers, not str
22.12.20 02:23:19 (+0000)  srly-ose-viewer  Traceback (most recent call last):
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "viewer.py", line 465, in <module>
22.12.20 02:23:19 (+0000)  srly-ose-viewer      main()
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "viewer.py", line 460, in main
22.12.20 02:23:19 (+0000)  srly-ose-viewer      asset_loop(scheduler)
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "viewer.py", line 317, in asset_loop
22.12.20 02:23:19 (+0000)  srly-ose-viewer      is_up_to_date()
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "/usr/src/app/lib/github.py", line 73, in is_up_to_date
22.12.20 02:23:19 (+0000)  srly-ose-viewer      latest_sha, retrieved_update = fetch_remote_hash()
22.12.20 02:23:19 (+0000)  srly-ose-viewer    File "/usr/src/app/lib/github.py", line 58, in fetch_remote_hash
22.12.20 02:23:19 (+0000)  srly-ose-viewer      latest_sha = resp.json()['object']['sha']
22.12.20 02:23:19 (+0000)  srly-ose-viewer  TypeError: list indices must be integers, not str
22.12.20 02:23:20 (+0000)  srly-ose-viewer  ./bin/start_viewer.sh: line 47: kill: (39) - No such process
22.12.20 02:23:23 (+0000)  srly-ose-viewer  chown: cannot access '/data/.screenly/latest_screenly_sha': No such file or directory
22.12.20 02:23:23 (+0000)  srly-ose-viewer  chown: cannot access '/etc/sudo.conf': No such file or directory
22.12.20 02:23:25 (+0000)  srly-ose-viewer  Loading browser...
22.12.20 02:23:27 (+0000)  srly-ose-viewer  Generating asset-list...
22.12.20 02:23:27 (+0000)  srly-ose-viewer  Current url is http://srly-ose-nginx:80/splash-page
22.12.20 02:24:27 (+0000)  srly-ose-viewer  Current url is http://srly-ose-nginx:80/static/img/loading.png
22.12.20 02:24:27 (+0000)  srly-ose-viewer  Viewer crashed.
22.12.20 02:24:27 (+0000)  srly-ose-viewer  Traceback (most recent call last):
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "viewer.py", line 465, in <module>
22.12.20 02:24:27 (+0000)  srly-ose-viewer      main()
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "viewer.py", line 460, in main
22.12.20 02:24:27 (+0000)  srly-ose-viewer      asset_loop(scheduler)
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "viewer.py", line 317, in asset_loop
22.12.20 02:24:27 (+0000)  srly-ose-viewer      is_up_to_date()
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "/usr/src/app/lib/github.py", line 73, in is_up_to_date
22.12.20 02:24:27 (+0000)  srly-ose-viewer      latest_sha, retrieved_update = fetch_remote_hash()
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "/usr/src/app/lib/github.py", line 58, in fetch_remote_hash
22.12.20 02:24:27 (+0000)  srly-ose-viewer      latest_sha = resp.json()['object']['sha']
22.12.20 02:24:27 (+0000)  srly-ose-viewer  TypeError: list indices must be integers, not str
22.12.20 02:24:27 (+0000)  srly-ose-viewer  Traceback (most recent call last):
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "viewer.py", line 465, in <module>
22.12.20 02:24:27 (+0000)  srly-ose-viewer      main()
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "viewer.py", line 460, in main
22.12.20 02:24:27 (+0000)  srly-ose-viewer      asset_loop(scheduler)
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "viewer.py", line 317, in asset_loop
22.12.20 02:24:27 (+0000)  srly-ose-viewer      is_up_to_date()
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "/usr/src/app/lib/github.py", line 73, in is_up_to_date
22.12.20 02:24:27 (+0000)  srly-ose-viewer      latest_sha, retrieved_update = fetch_remote_hash()
22.12.20 02:24:27 (+0000)  srly-ose-viewer    File "/usr/src/app/lib/github.py", line 58, in fetch_remote_hash
22.12.20 02:24:27 (+0000)  srly-ose-viewer      latest_sha = resp.json()['object']['sha']
22.12.20 02:24:27 (+0000)  srly-ose-viewer  TypeError: list indices must be integers, not str
22.12.20 02:24:28 (+0000)  srly-ose-viewer  ./bin/start_viewer.sh: line 47: kill: (37) - No such process
[... cut ...]
22.12.20 02:25:36 (+0000)  srly-ose-viewer  ./bin/start_viewer.sh: line 47: kill: (38) - No such process
22.12.20 02:25:39 (+0000)  srly-ose-viewer  chown: cannot access '/data/.screenly/latest_screenly_sha': No such file or directory
22.12.20 02:25:39 (+0000)  srly-ose-viewer  chown: cannot access '/etc/sudo.conf': No such file or directory
22.12.20 02:25:40 (+0000)  srly-ose-viewer  Loading browser...
22.12.20 02:25:42 (+0000)  srly-ose-viewer  Generating asset-list...
22.12.20 02:25:42 (+0000)  srly-ose-viewer  Current url is http://srly-ose-nginx:80/splash-page
```