Screenly/Anthias #1456: IP Address displays incorrectly on dashboard when running in balenaCloud

shawaj, 2020-12-20:
#### Overview of the Issue

When loading up Screenly OSE on a Pi 4 / Pi 400 using balenaCloud it displays the wrong IP address on the splash screen.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Set up screenly on balenaCloud using Pi4 or Pi 400
* Boot up device whilst connected to wlan or eth
* Check IP address...

[image]

[image]

#### Environment

* Raspberry Pi Hardware Version: Pi 4 or 400
* Raspberry Pi Network Setup: Wifi or ethernet
* Screenly OSE Version: latest from github

#### ifconfig output

```
root@162a2ea:~# ifconfig
balena0   Link encap:Ethernet  HWaddr 02:42:35:C5:B1:16  
          inet addr:10.114.101.1  Bcast:10.114.101.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

br-a39c8b498585 Link encap:Ethernet  HWaddr 02:42:D2:1C:6A:25  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:d2ff:fe1c:6a25/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1135 errors:0 dropped:0 overruns:0 frame:0
          TX packets:806 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1244054 (1.1 MiB)  TX bytes:81287 (79.3 KiB)

eth0      Link encap:Ethernet  HWaddr DC:A6:32:DB:FB:AF  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:7748 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7748 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:735606 (718.3 KiB)  TX bytes:735606 (718.3 KiB)

resin-dns Link encap:Ethernet  HWaddr FE:F0:EC:15:56:1B  
          inet addr:10.114.102.1  Bcast:0.0.0.0  Mask:255.255.255.0
          inet6 addr: fe80::fcf0:ecff:fe15:561b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:49 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:6707 (6.5 KiB)

resin-vpn Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.240.106.82  P-t-P:52.4.252.97  Mask:255.255.255.255
          inet6 addr: fe80::8a7c:93:1f8d:2f83/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:352 errors:0 dropped:0 overruns:0 frame:0
          TX packets:403 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:46392 (45.3 KiB)  TX bytes:51982 (50.7 KiB)

supervisor0 Link encap:Ethernet  HWaddr 02:42:2B:03:C4:F8  
          inet addr:10.114.104.1  Bcast:10.114.104.127  Mask:255.255.255.128
          inet6 addr: fe80::42:2bff:fe03:c4f8/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:38 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:5241 (5.1 KiB)

veth05c87f5 Link encap:Ethernet  HWaddr DE:AA:96:7F:01:C4  
          inet6 addr: fe80::dcaa:96ff:fe7f:1c4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8226 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9701 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:10561273 (10.0 MiB)  TX bytes:723245 (706.2 KiB)

veth30459fc Link encap:Ethernet  HWaddr 4A:32:32:48:64:A0  
          inet6 addr: fe80::4832:32ff:fe48:64a0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:525 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2836 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:38764 (37.8 KiB)  TX bytes:132841 (129.7 KiB)

veth5bd3b6d Link encap:Ethernet  HWaddr 16:49:8F:56:27:80  
          inet6 addr: fe80::1449:8fff:fe56:2780/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:49749 errors:0 dropped:0 overruns:0 frame:0
          TX packets:57868 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:11967859 (11.4 MiB)  TX bytes:11411906 (10.8 MiB)

veth6bf5ab0 Link encap:Ethernet  HWaddr 2A:08:C2:CA:51:30  
          inet6 addr: fe80::2808:c2ff:feca:5130/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7713 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8364 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:10597142 (10.1 MiB)  TX bytes:10622718 (10.1 MiB)

veth88b4bf7 Link encap:Ethernet  HWaddr D2:C0:B0:C3:9F:D3  
          inet6 addr: fe80::d0c0:b0ff:fec3:9fd3/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:69 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:9184 (8.9 KiB)

veth982451e Link encap:Ethernet  HWaddr 1E:CB:5D:57:AF:20  
          inet6 addr: fe80::1ccb:5dff:fe57:af20/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:55552 errors:0 dropped:0 overruns:0 frame:0
          TX packets:52063 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:11309093 (10.7 MiB)  TX bytes:12070541 (11.5 MiB)

wlan0     Link encap:Ethernet  HWaddr DC:A6:32:DB:FB:B0  
          inet addr:192.168.86.21  Bcast:192.168.86.255  Mask:255.255.255.0
          inet6 addr: fe80::3186:33b6:b8e4:916f/64 Scope:Link
          inet6 addr: fd14:c61e:9a88:1:5cb:140d:bc6:e0f0/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:89590 errors:0 dropped:462 overruns:0 frame:0
          TX packets:31967 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:76387831 (72.8 MiB)  TX bytes:6587141 (6.2 MiB)
```

As you can see - it is pulling the IP address of the `br-a39c8b498585` interface - instead of wlan0 which is the address I can access Screenly OSE on locally.

vpetersson, 2020-12-20:
This is known issue. It's very difficult to read the host IP from within a Docker container due to the network setup.

shawaj, 2020-12-20:
@vpetersson could you pull it in as an environment variable on first load of the container or something like that?

shawaj, 2020-12-20:
Something like this - https://nickjanetakis.com/blog/docker-tip-65-get-your-docker-hosts-ip-address-from-in-a-container

ealmonte32, 2020-12-21:
@shawaj 
I also looked at this method as potential way to do it, have you tested yourself by forking and editing the dockerfile?
I just started with testing the recent master branch and haven't gotten through list of pending items checklist..

shawaj, 2020-12-21:
@ealmonte32 haven't tested yet. But will do shortly.

And thanks for the tips re the wlan0 / eth0 that's really useful 😁

vpetersson, 2020-12-21:
I have looked into this issue in the past a number of times and it's tricky. You won't be able to get the IP of the host from within a container using tools like `ip` or similar. This is due to the fact that Docker does a lot of complex things with `iptables` and similar. 

There are a few possible solutions to this. 

* One is to use `host.docker.internal` as referenced in the article @shawaj posted. This solution however does not work on Docker on the Raspberry Pi unfortunately (even the latest version).
* Another solution is to pass it on as an environment variable from the host. This is the solution I was planning to implement, but it also comes with drawbacks (e.g. doesn't work on Balena and wouldn't work if the IP of the node changed after installation).

shawaj, 2020-12-22:
@vpetersson i was just looking through the balena docs and you could set the label for `io.balena.features.supervisor-api: '1'` and then use the following...

```
curl -X GET --header "Content-Type:application/json" \
    "$BALENA_SUPERVISOR_ADDRESS/v1/device?apikey=$BALENA_SUPERVISOR_API_KEY"
```

Output example:
```
{"api_port":48484,"ip_address":"192.168.816.221","os_version":"balenaOS 2.65.0+rev2","supervisor_version":"12.2.11","update_pending":false,"update_failed":false,"update_downloaded":false,"commit":"a4e498d2e28c90401b217f0d28e4b287","status":"Idle","download_progress":null}
```

So I guess you could just parse this JSON and grab the IP address from there?

closed 2020-12-22 by vpetersson

shawaj, 2020-12-22:
Just tested and working now...

[image]

commit 020d60c9 (2020-12-22): Resolves #1456 for Balena