Screenly/Anthias #1457: viewer container crash - No such file or directory & vlcpulse audio connection failure

LeSnowTiger, 2020-12-21:
#### Overview of the Issue

The ose viewer container crashes with the error message in the following log.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* `$ bash <(curl -sL https://www.screenly.io/install-ose.sh)`
* Select master / dev
* Select "no" for managing network
* Wait for installation / update
* Reboot

#### Environment

* Raspberry Pi Hardware Version: Pi 4 2GB
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: latest master / dev

#### Logs (docker logs DOCKERID)
```
chown: cannot access '/etc/sudo.conf': No such file or directory
Home directory not accessible: Permission denied
[01531510] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Home directory not accessible: Permission denied
[015422e8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Viewer crashed.
Traceback (most recent call last):
  File "viewer.py", line 529, in <module>
    main()
  File "viewer.py", line 494, in main
    setup()
  File "viewer.py", line 422, in setup
    db_conn = db.conn(settings['database'])
  File "/usr/src/app/lib/db.py", line 5, in <lambda>
    conn = lambda db: sqlite3.connect(db, detect_types=sqlite3.PARSE_DECLTYPES)
OperationalError: unable to open database file
Traceback (most recent call last):
  File "viewer.py", line 529, in <module>
    main()
  File "viewer.py", line 494, in main
    setup()
  File "viewer.py", line 422, in setup
    db_conn = db.conn(settings['database'])
  File "/usr/src/app/lib/db.py", line 5, in <lambda>
    conn = lambda db: sqlite3.connect(db, detect_types=sqlite3.PARSE_DECLTYPES)
sqlite3.OperationalError: unable to open database file
./bin/start_viewer.sh: line 47: kill: (37) - No such process
```

vpetersson, 2020-12-21:
@LeSnowTiger Thanks for reporting. Can you provide with the following information?

* `docker version`
* What Rasbian version you're running
* The result of `ls -lah /home/pi/.screenly`

LeSnowTiger, 2020-12-21:
@vpetersson 

- Docker Version: 19.03.13
- Raspbian Version: Raspbian GNU/Linux 10 (buster)

Result of `ls -lah /home/pi/.screenly`:
```
pi@iot00000002:~ $ ls -lah /home/pi/.screenly
total 48K
drwxr-xr-x  2 pi pi 4.0K Dec 21 09:28 .
drwxr-xr-x 10 pi pi 4.0K Dec  7 09:40 ..
-rw-r--r--  1 pi pi  12K Dec 21 12:43 celerybeat-schedule
-rw-r--r--  1 pi pi  266 Oct 21 11:34 default_assets.yml
-rw-r--r--  1 pi pi   15 Oct 21 12:14 device_id
-rw-r--r--  1 pi pi    0 Dec 21 09:28 initialized
-rw-r--r--  1 pi pi   41 Dec 21 12:47 latest_screenly_sha
-rw-r--r--  1 pi pi  600 Dec 21 09:06 screenly.conf
-rw-r--r--  1 pi pi 9.0K Dec  8 13:34 screenly.db
```

rusko124, 2020-12-21:
@vpetersson ignore the PulseAudio error. This error on initialization of VLC player and then we switch it to `alsa`.
https://github.com/Screenly/screenly-ose/blob/master/lib/media_player.py#L34

vpetersson, 2020-12-21:
Noted @rusko124. Thanks.

@LeSnowTiger so it seems like the issue is the viewer can't access the database. Could you try running the installer again for good measures and be on the lookout for errors?

LeSnowTiger, 2020-12-21:
Thanks for responding. I'cant see any errors while installing. I'll try to gather more information about the problem, but can't promise anything.

LeSnowTiger, 2020-12-21:
So I updated another raspberry. On the first run the following error appeard:

```
+ sudo -E docker-compose -f /home/pi/screenly/docker-compose.yml -f /home/pi/screenly/docker-compose.override.yml up -d
Creating network "screenly_default" with the default driver
Creating volume "screenly_screenly-data" with default driver
Creating volume "screenly_redis-data" with default driver
Creating volume "screenly_resin-data" with default driver
Creating screenly_redis_1           ... error
Creating screenly_srly-ose-server_1 ...
Creating screenly_srly-ose-server_1 ... error
ERROR: for screenly_srly-ose-server_1  Cannot start service srly-ose-server: failed to create endpoint screenly_srly-ose-server_1 on network screenly_default: failed to add the host (veth8c17f03) <=> sandbox (veth619d107) pair interfaces: operation not supported
ERROR: for redis  Cannot start service redis: failed to create endpoint screenly_redis_1 on network screenly_default: failed to add the host (veth22100ec) <=> sandbox (vethd580ae1) pair interfaces: operation not supported
ERROR: for srly-ose-server  Cannot start service srly-ose-server: failed to create endpoint screenly_srly-ose-server_1 on network screenly_default: failed to add the host (veth8c17f03) <=> sandbox (veth619d107) pair interfaces: operation not supported
ERROR: Encountered errors while bringing up the project.
```

No docker container was created / started.
After running the installation again the error did not appear a second time but the viewer container crashes with the same error from the top. Both raspberrys run with the same image. I think I'll try a clean installation tomorrow.

vpetersson, 2020-12-21:
@LeSnowTiger - My money would be on that it's a broken SD card. Basically, that sounds like a fairly low level docker issue and could be caused by read/write issues. Try a fresh Buster Light install on a fresh SD card.

ealmonte32, 2020-12-21:
I ran these same errors, both, re-running the installation fixed the docker container creation error, and as rusko said, the audio or cannot access home directory are temporary errors that go away after all containers are fully running..

LeSnowTiger, 2020-12-22:
Installing with a clean image fixed the above error. But I get a new one:
```
Traceback (most recent call last):
  File "viewer.py", line 465, in <module>
    main()
  File "viewer.py", line 460, in main
    asset_loop(scheduler)
  File "viewer.py", line 317, in asset_loop
    is_up_to_date()
  File "/usr/src/app/lib/github.py", line 73, in is_up_to_date
    latest_sha, retrieved_update = fetch_remote_hash()
  File "/usr/src/app/lib/github.py", line 50, in fetch_remote_hash
    'https://api.github.com/repos/screenly/screenly-ose/git/refs/heads/{}'.format(branch)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/adapters.py", line 514, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/screenly/screenly-ose/git/refs/heads/master (Caused by SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)'),))
./bin/start_viewer.sh: line 47: kill: (37) - No such process
```

But I'm really sure that this error is due to that the docker container is missing our root CA certificate. I'll have to talk to the network guy.

closed 2020-12-22 by LeSnowTiger