Screenly/Anthias #1459: Hitting GitHub rate limit

shawaj, 2020-12-21:
#### Overview of the Issue

Hitting github rest api rate limit when trying to deploy crashing containers.

#### Reproduction Steps

Steps to reproduce this issue:

* clone latest repo
* balena push
* wait for containers to load
* viewer and server containers crash and after a while hit the github rate limit

#### Environment

* Raspberry Pi Hardware Version: Pi 400
* Raspberry Pi Network Setup: WiFi
* Screenly OSE Version: lates from GitHub

#### Error message from GitHub

```
Invalid response from GitHub: {"message":"API rate limit exceeded for xx.163.xxx.xx. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)","documentation_url":"https://developer.github.com/v3/#rate-limiting"}
```

shawaj, 2020-12-21:
FYI I presume this is a very unlikely issue for anyone to experience in practice and is likely just because I have been pushing containers to multiple devices to test the implementation with balenaCloud and so have run up against the API limit.

Not sure if there is a way to authenticate the request to remove the rate limit like it says?

ealmonte32, 2020-12-22:
I am also getting crashing containers, I have been testing on Pi4, the latest pull was cloned like around 10PM Eastern..

do your logs look similar (i know you're running balena, but journald should show same logs I believe) ? https://pastebin.com/T3nfcpuW

And this was not happening from the master branch cloned yesterday..

vpetersson, 2020-12-22:
Interesting. It should only check every 24h. It should be cached in Redis. Can you both try to pull down the latest update from master updates?

shawaj, 2020-12-22:
Not sure if it is relavent - but I am still getting this issue in viewer:

```
22.12.20 16:58:25 (+0000)  srly-ose-viewer  Unable to get Git branch
22.12.20 16:58:25 (+0000)  srly-ose-viewer  Unable to get latest version from GitHub
```

Apart frmo that everything seems to be working nicely now 👍

vpetersson, 2020-12-22:
Excellent! Yeah, this is related to [this issue](https://forums.balena.io/t/git-push-and-buildargs/227059/2). Feel free to ping anyone internally to see if we can find a solution to this as it is Balena specific.

vpetersson, 2020-12-30:
Closing this out as it should now have been resolved.

closed 2020-12-30 by vpetersson