Screenly/Anthias #1470: Bad character escape

ArcasCZ, 2021-01-09:
#### Overview of the Issue

When I add any URL containing more than one GET parameter separated by & (ampersand), it's replaced by &amp and it does not work as it should. I think it's 'cause of bad character escaping - it encode characters to store the URL, but doesn't decode it afterwards when loading asset.

#### Reproduction Steps

* Press Add Asset
* Enter the Asset URL https://weather.srly.io?lat=50.11419438385983&lng=17.696658078595718&lang=cs&24h=1
* Change Play for to Forever, enable this asset
* Wait for the asset to show up on the screen
* It should show location Krnov, but it shows location by your IP

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 Model B; 4GB
* Screenly OSE Version: production@d12df55

vpetersson, 2021-01-09:
Try upgrading to the developer version as I believe this has been resolved there.

ArcasCZ, 2021-01-10:
Tried, now Screenly won't even start...

vpetersson, 2021-01-11:
We need more information in order to help you. Please provide full logs. Otherwise we can't do anything.

ArcasCZ, 2021-01-12:
I'm gonna do full reinstall first since it's faster (no personal settings yet). If it won't work, I'll send you full logs.

Also, this command is something I was looking for...

vpetersson, 2021-01-12:
Sounds good.

ArcasCZ, 2021-01-12:
Creating screenly_srly-ose-server_1 ... error
Creating screenly_redis_1           ...

ERROR: for screenly_srly-ose-server_1  Cannot start service srly-ose-server: failed to create endpoint screenly_srly-ose-server_1 on network screenly_default: failed to add the host (vethddf52b2) <=> sandbox (vethdcf19a1) pair interfacesCreating screenly_redis_1           ... error

ERROR: for screenly_redis_1  Cannot start service redis: failed to create endpoint screenly_redis_1 on network screenly_default: failed to add the host (veth3447b4d) <=> sandbox (veth41d3989) pair interfaces: operation not supported

ERROR: for redis  Cannot start service redis: failed to create endpoint screenly_redis_1 on network screenly_default: failed to add the host (veth3447b4d) <=> sandbox (veth41d3989) pair interfaces: operation not supported

ERROR: for srly-ose-server  Cannot start service srly-ose-server: failed to create endpoint screenly_srly-ose-server_1 on network screenly_default: failed to add the host (vethddf52b2) <=> sandbox (vethdcf19a1) pair interfaces: operation not supported
ERROR: Encountered errors while bringing up the project.

vpetersson, 2021-01-12:
Hmm. Is this a clean installation? Seems like something is wrong with Docker.

ArcasCZ, 2021-01-12:
This is what I found when I was doing clean install in terminal.

vpetersson, 2021-01-12:
From experience, when I see odd errors like this, I always suspect two things:

* A broken SD card
* Insufficient power supply (i.e. not using the official one)

Can you try swapping out the SD card and try with an official power supply (if you're not using it already)?

ArcasCZ, 2021-01-12:
I'll swap the card and let you know.

ArcasCZ, 2021-01-12:
Another SD card, install ended correctly, still not working. Command output is the same.

Also, I think I forgot to expand root file system with the first SD card.

ArcasCZ, 2021-01-12:
Closing for now, I have problem setting up headless Screenly. Will tell if this won't work

vpetersson, 2021-01-12:
It's strange. I just did a fresh install just last night on the latest Raspbian/Raspberry Pi OS (Lite) without any issues.

ArcasCZ, 2021-01-12:
IDK, now it works, but I'm not able to get to System info, as it ends with 504. And VNC is not working at all (Cannot currently show the desktop)

closed 2021-01-12 by ArcasCZ

vpetersson, 2021-01-12:
> IDK, now it works, but I'm not able to get to System info, as it ends with 504.

This is expected. I am working on refactoring this.

> And VNC is not working at all (Cannot currently show the desktop)

This is also expected. You can't use it like that. If you're having X11/Xorg installed, it will likely cause issues anyways. Screenly OSE renders directly to framebuffer, so there is no need for X11/Xorg. You *might* be able to get it working using something like [this](https://github.com/ponty/framebuffer-vncserver) but that's entirely untested.

ArcasCZ, 2021-01-12:
Works like a charm, thanks!