Screenly/Anthias #1471: Celery container crashing

vpetersson, 2021-01-12:
#### Overview of the Issue

It appears as the celery container crashes on master

Logs:

```
$ docker logs screenly_srly-ose-celery_1
* failed to open vchiq instance
* failed to open vchiq instance
* failed to open vchiq instance
* failed to open vchiq instance
* failed to open vchiq instance
* failed to open vchiq instance
* failed to open vchiq instance
* failed to open vchiq instance
* failed to open vchiq instance
[...]
```

#### Reproduction Steps

Unclear atm

#### Environment

* Raspberry Pi Hardware Version: 3 Model B
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter): WiFi
* Screenly OSE Version: Latest on master

ealmonte32, 2021-01-21:
Happening to me as well now troubleshooting with Pi4.

ealmonte32, 2021-01-21:
Fix for it found, dealing with devices for dev/vchiq not set on dockerfile for celery but instead on viewer.

ealmonte32, 2021-01-26:
this can be closed, fixed on https://github.com/Screenly/screenly-ose/pull/1477

closed 2021-01-27 by vpetersson