Screenly/Anthias #1472: API Docs does not working

viradhanus, 2021-01-13:
[API Docs](https://ose.demo.screenlyapp.com/api/docs/) in README.md does not load.

vpetersson, 2021-01-13:
This should be working now. Had to update it based on the recent developments. However, you can always access the API docs on your local Pi.

closed 2021-01-13 by vpetersson

viradhanus, 2021-01-14:
@vpetersson Thank you... But still it gives 502 error.. hopefully you guys will fix it soon.
[image]

vpetersson, 2021-01-14:
The demo site is using the developer version and it's under active development. I just resolved it. Tl;dr, it had to do with docker images of various architectures clashing.