Screenly/Anthias #1473: Stuck - Asset in Processing

nicholas-sousa, 2021-01-13:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A youtube url was placed for testing, and it got stuck in the inactive and not enabling to delete it

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL https://www.youtube.com/watch?v=HSsqzzuGTPo
* Press "On"
* Wait for the asset to show up on the screen

#### Environment

* Raspberry Pi 3 Hardware Version: 1.2
* Raspberry Pi Network Setup: ethernet
* Screenly OSE Version: latest

vpetersson, 2021-01-14:
When you say latest, do you mean the latest disk image, or the developer version? If the former, then please try the following:

* Download the latest Raspbian/Raspberry Pi Lite image and using ([Raspberry Pi Imager](https://www.raspberrypi.org/software/).
* Enable SSH and then follow the [installation instructions](https://github.com/Screenly/screenly-ose#installing-on-raspbianraspberry-pi-os). Make sure to select the 'Development/Master branch'.

Let us know if that does/doesn't solve the issue.

rusko124, 2021-01-27:
@vpetersson @nicholas-sousa
It's not stuck. When you add a youtube video as an asset it means that the video will be downloaded. Your asset is a 3hour of 4k video. Thus you just need to wait when your device will download the video.

closed 2021-01-27 by vpetersson

semaf, 2022-05-27:
> @vpetersson @nicholas-sousa I guess It's not stuck. When you add a youtube video as an asset it means that the video will be downloaded. Your asset is a 3hour of 4k video. Thus you just need to wait when your device will download the video.

That will be a problem if a livestream is running, like a radio channel.

vpetersson, 2022-05-30:
Yes, live streams are not supported. The current implementation will pull down and cache the video file. Hence a live stream would not work.