Screenly/Anthias #1475: Error with install

Goteez, 2021-01-19:
Running RPI3, upgrading existing screenly ose with the command:

bash <(curl -sL https://www.screenly.io/install-ose.sh)

I get this near the end:

TASK [screenly : Create screenly-server container] **************************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Error pulling image screenly/screenly-ose-server:latest - 404 Client Error: Not Found (\"{\"message\":\"pull access denied for screenly/screenly-ose-server, repository does not exist or may require 'docker login': denied: requested access to the resource is denied\"}\")"}

vpetersson, 2021-01-20:
@Goteez That's strange. I don't really see any references to this in the code base. Did you select the production version? If so, try switching to the developer version as that's where all work is going into at the moment.

Goteez, 2021-01-20:
I should have mentioned that I was using the developer version.

vpetersson, 2021-01-20:
That's very odd. There is no reference to this in the master branch. Maybe something went wrong. Try this:

```
$ cd /home/pi/screenly
$ git checkout master
$ git pull
$ ./bin/run_upgrade.sh
```

Goteez, 2021-01-21:
> CAUTION: This message was sent from outside the organization. Please do not click links, open attachments, or respond unless you recognize the source of this email and know the content is safe.

I finally had a chance to run your suggestion today. The git pull sync’d a lot of files. I ran the upgrade as requested and did not get that error. It appears to have completed fine.

Thanks.

Ryan Bohn
Network and Systems Analyst
Regional District of Central Okanagan
250-469-6273 | ryan.bohn@rdco.com<mailto:ryan.bohn@rdco.com>
Connect with us | rdco.com

Goteez, 2021-01-21:
I spoke too soon, all I get now is the screenly logo on the display, and the web page doesn't respond at all. Looks like its busted. Is it better that I just start from scratch here?

Attached is the upgrade log.
[upgrade.txt](https://github.com/Screenly/screenly-ose/files/5851843/upgrade.txt)

ealmonte32, 2021-01-21:
> I spoke too soon, all I get now is the screenly logo on the display, and the web page doesn't respond at all. Looks like its busted. Is it better that I just start from scratch here?
> 
> Attached is the upgrade log.
> [upgrade.txt](https://github.com/Screenly/screenly-ose/files/5851843/upgrade.txt)

I suggest you clear the screenly local files unless you were manually editing the code or doing some local development on files under the `./screenly/` directory, if not, simply remove the directory: `rm -rf /home/pi/screenly` which is the same as `./screenly` if you are logged in as Pi user and under the Pi home directory.
Then, reinstall everything by pulling all latest files by simply running: `bash <(curl -sL https://www.screenly.io/install-ose.sh`
and then pick option 2, and maybe say yes to updating system since it seems you might have outdated packages..

Goteez, 2021-01-21:
OK, thanks for the tip. I deleted the screenly directories, then ran the install fresh, and I am left with the same screenly logo on the display and no web interface. I had already updated the system on a previous attempt, the first one actually.

Attached is the log of the install.
[install.txt](https://github.com/Screenly/screenly-ose/files/5852501/install.txt)

ealmonte32, 2021-01-21:
Ok, so it's probably a package/module issue dealing with docker compose, can you paste the result of the following two commands separately:
`pip list`
`dpkg -l | grep -i docker`

Goteez, 2021-01-21:
```
pi@klo-foyer-tv:~ $ pip list
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Package                            Version
---------------------------------- -----------
ansible                            2.8.8
asn1crypto                         0.24.0
attrs                              20.3.0
backports.shutil-get-terminal-size 1.0.0
backports.ssl-match-hostname       3.7.0.1
cached-property                    1.5.2
certifi                            2019.6.16
cffi                               1.12.3
chardet                            3.0.4
configparser                       4.0.2
contextlib2                        0.6.0.post1
cryptography                       2.7
distro                             1.5.0
docker                             4.3.1
docker-compose                     1.26.2
docker-py                          1.10.6
docker-pycreds                     0.4.0
dockerpty                          0.4.1
docopt                             0.6.2
enum34                             1.1.6
functools32                        3.2.3.post2
idna                               2.8
importlib-metadata                 2.1.1
ipaddress                          1.0.22
Jinja2                             2.10.1
jsonschema                         3.2.0
MarkupSafe                         1.1.1
netifaces                          0.10.4
pathlib2                           2.3.5
pip                                20.3.3
pwgen                              0.8.2.post0
pycparser                          2.19
pyrsistent                         0.16.1
python-apt                         1.4.3
python-dotenv                      0.15.0
PyYAML                             5.1.1
requests                           2.22.0
RPi.GPIO                           0.6.5
scandir                            1.10.0
setuptools                         41.0.1
sh                                 1.8
six                                1.12.0
subprocess32                       3.5.4
texttable                          1.6.3
typing                             3.7.4.3
urllib3                            1.25.3
websocket-client                   0.56.0
wheel                              0.33.4
zipp                               1.2.0
pi@klo-foyer-tv:~ $ dpkg -l | grep -i docker
ii  docker                          1.5-1                             armhf        System tray for KDE3/GNOME2 docklet applications
ii  docker-ce                       5:19.03.13~3-0~raspbian-stretch   armhf        Docker: the open-source application container engine
ii  docker-ce-cli                   5:19.03.13~3-0~raspbian-stretch   armhf        Docker CLI: the open-source application container engine
```

ealmonte32, 2021-01-21:
There it is, you are not running the system with a fresh Raspbian install, I assume you were using it for other stuff or installing and testing other programs, uninstall the following:
`sudo pip uninstall docker-py docker-pycreds`
`sudo apt purge docker` <-- not sure if this one might need to be removed but try with removing the pip ones first, then when its done, reboot.. i dont think it will be necessary to recreate the containers but if necessary we can easily do so with a few commands, so try this first.

Goteez, 2021-01-21:
HI there, ya i am reusing the same raspbian where screenly was installed but broke. I don't use it for anything other then screenly. I will run this and report back.

ealmonte32, 2021-01-21:
Since you are going to use it for screenly only, I would suggest you clear that SD card and install the latest raspian lite instead for a best performing situation, like even if the above commands make it work or if you have to recreate the containers to make it work, I think if you have some time, its best to do a fresh clean install by getting the buster lite image from raspberry repo: 
http://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-01-12/

and then running the bash install script and that way you get a 100% fresh install system with only the required packages, etc, guaranteed to work (except for any currently pending issues that affect the Master/Dev branch if you select it).

Goteez, 2021-01-21:
Ya, you have the right idea. It still isnt working, so clean slate is the way to go. Does it work with buster now?

ealmonte32, 2021-01-21:
I know that if we recreate the containers it would work because those docker python packages you had are the culprit since I was reading on it on another github page, but rather than rebuilding the containers etc, yeah just do the fresh install better.. 
yes buster has worked for a while now, that URL i gave you includes the latest raspbian os lite for Pi3, download the zip file, and use balenaEtcher are you normally would..
remember if you need the production version to select option 1 because current master branch has some tiny bugs that we still need to iron out, but you could use it if you dont mind dealing with the current bugs which arent that big of a deal in general and you at least get the new WebView browser which makes all modern websites load unlike the uzbl in production..

Goteez, 2021-01-22:
Thanks very much. I have put the latest raspbian buster image on my sd card and then installed the latest screenly development version and all is working now. Perhaps my issues were the result of running raspbian stretch, and trying to upgrade an almost 2 year old screely ose install. Thanks again.

ealmonte32, 2021-01-22:
No prob.. yes that was probably the reason.
Can you close this issue so that we clear some of these pending ones..
thanks

closed 2021-01-22 by Goteez