Screenly/Anthias #1480: relying on get-pip.py vs python-pip

ealmonte32, 2021-01-25:
tried to install screenly on fresh buster image and ran into this error today:

```
+ curl -s https://bootstrap.pypa.io/get-pip.py
+ sudo python
Traceback (most recent call last):
  File "<stdin>", line 24226, in <module>
  File "<stdin>", line 199, in main
  File "<stdin>", line 82, in bootstrap
  File "/tmp/tmp_Fskdo/pip.zip/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
```

first I checked on install.sh to see if anything changed, and wasnt that, then went to get-pip.py github account to see if it was something on their end, and it was: https://github.com/pypa/get-pip/issues/83

I think that we should try and not rely on that get-pip.py installation when we can simply install it from debian repo..
`sudo apt install python-pip --no-install-recommends`

Simply because we get the necessary packages we need to run pip installations...

@vpetersson , thoughts?

#### Environment

* Raspberry Pi Hardware Version: Pi4

vpetersson, 2021-01-25:
Yeah, I expected something like this would happen now that [python 2 support has been removed](https://pip.pypa.io/en/stable/news/#id1).

The best path forward is likely what you are suggesting: using distribution packages instead.

ealmonte32, 2021-01-25:
Sounds good.. I see you made the commit, I'll close this issue.

closed 2021-01-25 by ealmonte32