Screenly/Anthias #1484: viewer memory consumption on development branch

ealmonte32, 2021-01-27:
The high memory consumption issue we used to experience or still experience on production branch due to `uzbl` is seen now on the development branch with docker as well. I have only experienced this the most on Pi3 hardware due to the total usable memory of less than ~800MB.

The container memory limit needs to take into account the memory being consumed by other services/processes as well, thus I think we need to maybe look further into this, or maybe 70% is the best balance of enough to allow for responsiveness while also allowing for container to not crash before it reaches a max..

I have yet to see the cgroup limits start OOM and kill the container, which I assume is because the limit hasn't been reached, but the device still becomes very laggy, which is better than unresponsive, although before your 70% memory limit I did get the Pi3 to become unresponsive during testing.

I will continue to try and stress test the Pi3 with many heavy assets to see if cgroups/oom are working properly to not allow the Pi to become unresponsive as it was doing before the memory limits were set.

#### Environment

* Raspberry Pi Hardware Version: Pi3
* Screenly OSE Version: master/dev branch

---

Just some screenshots here for reference:

[image]

[image]

vpetersson, 2021-01-28:
Good findings, @ealmonte32. Yeah the current memory limit was just a guesstimate. It is very possible that we will need to tweak it a bit more by lowering it. The downside with that of course is that some web pages won't render then.

nkaminski, 2021-09-23:
I'm encountering this exact same issue and was able to prevent the pi from crashing by lowering the viewer memory limit to (system memory - 512MB). 

This now causes the OOM killer to trigger every ~36hrs, however the OOM killer opts to kill the renderer processes instead of the init process of the viewer container, resulting in a responsive Pi but with a persistently blank white display.

Therefore, I am currently testing with the following line in the start_viewer.sh script immediately before the wait-for-exit loop such that the OOM killer kills the container init script (triggering a restart of the container) instead of a random renderer proc:

`echo 1000 > /proc/$$/oom_score_adj`

closed 2022-01-09 by stale[bot]