Screenly/Anthias #1485: Question about customized user's graphics display

HamzaHajeir, 2021-01-28:
Hi

I’m in development of my first product in market. I’m looking for an efficient way to display Ads and device’s interactive graphics in interoperable way.

The device measures Human weight/height on coin insert, displays the results to the display. which is currently ‘7 segments display’.

My application is running on ESP32,

However, I’m investigating how can I add display to the device running adds, which could be interrupted from the application MCU to display interactive graphics/texts (weight / height).

The MCU could send message/command through I2C/SPI/UART to RPI.
RPI can read this message.

But I don’t know how to complete the cycle with display/Screenly.

Let’s summarize it by a sequence :

- Display is displaying ads.
- Human measures weight/height.
- A command to display weight/height that is being read by RPI.
- RPI decides to gently shut off ads, And display some graphics that’s related to the data/command came from MCU.
- RPI returns playing Ads.
And All of this need to be done locally, with no requirement of Internet access to accomplish this.

Any thoughts?

vpetersson, 2021-01-29:
> I’m in development of my first product in market.

Exciting. Congrats.

> The MCU could send message/command through I2C/SPI/UART to RPI.

Sounds doable in theory, but you wouldn't be able to use I2C/SPI/UART, but rather the [API](https://ose.demo.screenlyapp.com/api/docs/). You'd have to basically write some logic around this. If you're running on an ESP32, there shouldn't be any issues firing off some API calls against the Pi.

> And All of this need to be done locally, with no requirement of Internet access to accomplish this.

In that case, you will need to add additional hardware (an RTC). Otherwise, the Raspberry Pi will not know the time after a reboot. 

But yes, in theory this should be possible. 

If you need to push some kind of payload (in addition to the event) from the ESP32, it might be easier to write some kind of webapp/service on the Pi that you interface with from the ESP32. This webapp/service would then in turn interface with Screenly OSE and display say a webpage with the payload received from the ESP32.

Hopefully this makes sense.

closed 2021-01-29 by vpetersson

HamzaHajeir, 2021-01-31:
Thanks @vpetersson for your informative and encouraging comment.

I've taken a quick look to APIs.
However, I doubt I've explained the purpose correctly. I need the both features together : Sync with assets through a server to display ads, and Interrupt it locally with payload.

> If you need to push some kind of payload (in addition to the event) from the ESP32, it might be easier to write some kind of webapp/service on the Pi that you interface with from the ESP32. This [...]

We're in agree that some script/service would be needed on the Pi. 

Where can I read interfacing with Screenly APIs/Docs from this kind of service ?

ealmonte32, 2021-01-31:
>Where can I read interfacing with Screenly APIs/Docs from this kind of service ?

https://ose.demo.screenlyapp.com/api/docs/