Screenly/Anthias #1486: Upload not working for mp4 files

MangNPB, 2021-01-30:
First off thank you very much to quick response on fixing the previous issue with uploads. 

I update SOMO again by reinstalling both SOMO and SOMA. Upload for jpg works as expected. The problem arise on uploading MP4 files using player's add asset button and multiplayer upload. On singler player upload upload window abruptly closed when it seems to be done but no new asset shown on player's list on SOMO nor Screenly ose web ui. Multiplayer upload shows reload button when completed and still no new asset for mp4 files. Only JPG went through perfectly.

Mang

vpetersson, 2021-02-01:
These projects are not affiliated with us. You need to raise this with the maintainer of those projects.

vpetersson, 2021-02-01:
If you are able to reproduce it on a single device using the web interface directly, then please provide logs.

MangNPB, 2021-02-01:
It's already solved by your mate.  Thanks

closed 2021-02-01 by MangNPB