Screenly/Anthias #1487: Installation not work via script!

didiatworkz, 2021-01-31:
#### Overview of the Issue

``` 
++ curl -s https://raw.githubusercontent.com/Screenly/screenly-ose/production/requirements/requirements.host.txt
++ grep ansible
+ ANSIBLE_VERSION=
+ sudo pip install ''
Invalid requirement: ''
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/req/req_install.py", line 82, in __init__
    req = Requirement(req)
  File "/usr/share/python-wheels/packaging-16.8-py2.py3-none-any.whl/packaging/requirements.py", line 94, in __init__
    requirement_string[e.loc:e.loc + 8]))
InvalidRequirement: Invalid requirement, parse error at "''"

+ sudo -u pi ansible localhost -m git -a 'repo=https://github.com/screenly/screenly-ose.git dest=/home/pi/screenly version=production force=no'
sudo: ansible: command not found
```

The file: ´http://raw.githubusercontent.com/Screenly/screenly-ose/production/requirements/requirements.host.txt´ doesn't exist!

#### Reproduction Steps

Steps to reproduce this issue, eg:

Installation Screenly OSE via command: `bash <(curl -sL https://www.screenly.io/install-ose.sh)`
Choose: 
+ (1) = production
+ y = network management
+ y = system upgrade

#### Environment

* Raspberry Pi Hardware Version: Pi3 B+ -- Linux raspberrypi 4.14.98-v7+ (stretch)
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: 0.18.3

vpetersson, 2021-01-31:
I've pushed a fix for this. Try again please.

closed 2021-01-31 by vpetersson

didiatworkz, 2021-01-31:
Thx for the quick fix :)

commit b562115c (2021-01-31): Closes #1487. Fixes logic for production install