Screenly/Anthias #1490: Backup/Restore Still Not Working with Balena Cloud Install

TheQuank, 2021-02-07:
Howdy everyone! I've done quite a bit of reading about Backup/Restoring in Screenly when installed with Balena Cloud. There seems to be a number of differing options about it working/not working and whether or not its fixed. I've tried with clean installs over the last couple of days and am not able to get it to work. 

Error Message is: Server Error: [Errno 2] No such file or directory: '/data/screenly/static/screenly-backup-2021-02-07T18-26-10.tar.gz'

[image]

I'm running this on balenaOS 2.54.2+rev1 the latest (as of 2/7/21) stable version of Screenly. Any help would be much appreciated!!

TheQuank, 2021-02-09:
So if I manually create the /data/screenly/static path with 777 permissions inside the srly-ose-server container, it appears to generate the backup file, then errors when trying to download it:

Sorry, this page does not exist!

Logs: 09.02.21 06:58:57 (-0600)  srly-ose-nginx  192.168.5.136 - - [09/Feb/2021:12:58:57 +0000] "GET /favicon.ico HTTP/1.1" 200 32 "http://192.168.5.206/static_with_mime/Reception%20Screenly-2021-02-09T12-56-25.tar.gz?mime=application/x-tgz" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.150 Safari/537.36 Edg/88.0.705.63" "-"

ealmonte32, 2021-02-10:
@TheQuank 
hmm.. maybe this was forgotten in the QA checklist..
I am able to reproduce this even on native non-balena master branch (this is what we call the version you're referring to)..
I will do some troubleshooting, hoping this is most likely a simple fix on the Dockerfile/docker-compose to create those directories as persistent storage and not manually creating it after the fact..
Will post back if I figure anything out.. if you play around with the Dockerfile and docker-compose files and get it to work first, just post here as well.
Thx

ealmonte32, 2021-02-10:
Ok I fixed it locally, needed to add the alias in nginx.. making PR for it now..

ealmonte32, 2021-02-10:
@TheQuank pls run the bash script again to reinstall and it should work.
Dont forget to clone/pull the github repo again since two files were modified and these need to be updated in your local directory when rebuilding the image for balena.

BrendonKoz, 2021-05-20:
@ealmonte32 Using a pi 3b: Fresh OS [via raspberry pi imager] and Screenly install today using `bash <(curl -sL https://www.screenly.io/install-ose.sh)`, choosing option 2 (Development/Master version), not letting Screenly manage the network, and running a full-upgrade (even though the raspi host was fully up-to-date). I still cannot use the UI on a non-balena setup to initiate a backup. I get a **403 Forbidden** error reported from nginx. Is this possibly related as a regression error?

BrendonKoz, 2021-05-20:
Assuming this is up-to-date, I just saw [the roadmap document](https://github.com/Screenly/screenly-ose/blob/master/docs/roadmap.md#known-issues-in-master-branch) which already mentions this as not yet working. If the roadmap document is up-to-date, please disregard. :)

ealmonte32, 2021-05-21:
@BrendonKoz 
Hmm.. not sure, have to test myself and get back to you.
Been busy away from OSE, but will try to test on one pi3b and see if I can resolve.

ealmonte32, 2021-05-25:
@BrendonKoz 
Tested this just now, installed fresh raspian OS, bash install option 2, etc etc.. add some assets to test, go to settings, get backup and i get the tar.gz file.. i forgot since its been a while working with this but if you need assets like URLs and websites etc (since the backup only includes files uploaded), you can simply use the API URL at `http://192.168.1.14/api/v1.1/assets` and get the json contents of the assets, then maybe beautify it with website like `https://codebeautify.org/jsonviewer` just to have a backup of URL assets.

BrendonKoz, 2021-06-02:
I've unfortunately moved to a Balena install, so the scenario is a little different, but I'm still getting an error (clicking on "Get Backup": `Server Error: [Errno 2] No such file or directory`). I will spin up another version to double-check, but I think I may be getting an error because all of my _active_ assets are currently URLs. I do have inactive assets that are images. The API does properly display a JSON view of the assets correctly, however...and if need be I can always copy the files, including the screenly.db, directly off the device to get a backup.

BrendonKoz, 2021-06-04:
Tested the basic installation's backup and this time it worked. Not sure why or how, but we'll chalk it up to coincidence. Balena backup still seems to fail though, as this issue states. Thanks for testing it yourself, and making me look at it again, @ealmonte32! Sorry for the busy work, but glad it was currently verified to be OK. Thank you.

ealmonte32, 2021-06-04:
Hello,
When you say "basic installation's backup", do you mean you chose the Production version (option 1) of Screenly-OSE vs the development version (option 2)? because that would make sense since the current production does not use docker containers.

BrendonKoz, 2021-06-04:
Sorry, the bash-command-based non-Balena installation. I chose the development version in my test today where the backup worked. So...
Installed fresh raspbian. Install option 2. Don't configure network. Do a system upgrade. etc. I apologize for not being more clear.

ealmonte32, 2021-06-08:
It's ok, I get it now, the issue seems to be with balena installations only then..

closed 2022-04-16 by stale[bot]