Screenly/Anthias #1491: Screen going black

shaqaruden, 2021-02-09:
After an unknown period of time device goes blank. If I restart the screen-viewer service it comes back up again.

#### Environment

* Raspberry Pi Hardware Version: RPI4
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter) Built-in Wifi
* Screenly OSE Version: production@d12df55

ealmonte32, 2021-02-10:
This is probably due to the memory consumption + uzbl = oom_killer taking effect.

You would either need to upgrade to master/dev branch, or set a python viewer memory limit hard-coded into the program such as this workaround I had done on a testing version of mine:
```
#memory consumption fix for viewer.py

#import the necessary modules
import resource
from psutil import virtual_memory

#this would be the base limits of ~600MB and 700MB as (soft,hard) limits
resource.setrlimit(resource.RLIMIT_AS, (600000000,700000000))

#this is the code that gets it done

mem = virtual_memory() #assigns all the results of the current system memory to the mem variable
mem_soft = (mem.total * .75) #the mem variable has total which ouputs memory as integer, this assigns mem_soft a value of 75% of it
mem_hard = (mem.total * .90) #the mem variable has total which ouputs memory as integer, this assigns mem_hard a value of 90% of it

num_format = "{:.0f}".format #this is to format the results into a 9 digit number

#this is what combines the numbers retrieved from above and sets the full command as combined
setlimit = 'resource.setrlimit(resource.RLIMIT_AS, (' + num_format(mem_soft) + ',' + num_format(mem_hard) + '))'
exec(setlimit) #this executes the full command

#this is so that every system can log the current limits upon the start of the viewer
logging.info("The current resource limit is set as:")
logging.info(setlimit)
```

shaqaruden, 2021-02-10:
I'll look into this and get back to you

shaqaruden, 2021-02-11:
I only have 1GB models right but the this does not seem to be caused by memory usage.
[image]

shaqaruden, 2021-02-11:
Heres the full htop screenshot

[image]

shaqaruden, 2021-02-11:
Looking at journelctl I am running out of memory

> Feb 09 19:56:47 RPI58 kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=uzbl-core,pid=695,uid=1000

> Feb 09 19:56:47 RPI58 kernel: Out of memory: Killed process 695 (uzbl-core) total-vm:958748kB, anon-rss:402536kB, file-rss:0kB, shmem-rss:48612kB, UID:1000 pgtables:1260kB oom_score_adj:0

shaqaruden, 2021-02-11:
@ealmonte32 where do I add that code? viewer.py?

ealmonte32, 2021-02-11:
> @ealmonte32 where do I add that code? viewer.py? If so does it need to anywhere in particular within that file?

Yes in `viewer.py`, and well I set it up somewhere in the top when I was using it.. the importing of modules in their proper areas and then somewhere in the very top right after all modules were imported.

Dont forget to play with the soft and hard limit as those are in Bytes, so I would say that with a 1GB Pi you could maybe set soft to 600000000 and hard to 700000000 just because you need to leave some memory for every other process..

Also, keep in mind you are selecting the production version which does not have all the fixes such as the very important one of `&amp` fix in the URI. You would need to edit your `server.py` for this part and add only the commits for the ampfix which would be for example on the file: https://github.com/Screenly/screenly-ose/blob/master/server.py
```
starting in line 489:
    ampfix = "&amp;"
    name = escape(get('name').replace(ampfix, '&'))
.
.
.
then further down on 511:
    uri = (get('uri')).replace(ampfix, '&').replace('<', '&lt;').replace('>', '&gt;').replace('\'', '&apos;').replace('\"', '&quot;')

```

you would need to make those changes in your production version of the server.py file in order for you to have proper URL names and symbols accepted into the URI.

shaqaruden, 2021-02-17:
Thanks, I modified viewer.py and the board I tested it on has ran for almost a week now without blanking

shaqaruden, 2021-02-23:
I have been updating viewer.py on all my pis. I have gotten to one where it breaks it. I get the following error in journalctl
```
Feb 22 20:41:38 RPI63 systemd[1]: Starting Screenly Viewer...
Feb 22 20:41:39 RPI63 xset[856]: server does not have extension for -dpms option
Feb 22 20:41:39 RPI63 systemd[1]: Started Screenly Viewer.
Feb 22 20:41:40 RPI63 python[858]: Traceback (most recent call last):
Feb 22 20:41:40 RPI63 python[858]:   File "/home/pi/screenly/viewer.py", line 52, in <module>
Feb 22 20:41:40 RPI63 python[858]:     exec(setlimit) #this executes the full command
Feb 22 20:41:40 RPI63 python[858]:   File "<string>", line 1, in <module>
Feb 22 20:41:40 RPI63 python[858]: ValueError: not allowed to raise maximum limit
```

Here is the code I added
```python
#memory consumption fix for viewer.py

#import the necessary modules
import resource
from psutil import virtual_memory

#this would be the base limits of ~3600MB and 3700MB as (soft,hard) limits, you can play around with these values to fit your needs
resource.setrlimit(resource.RLIMIT_AS, (600000000,700000000))

#this is the code that gets it done

mem = virtual_memory() #assigns all the results of the current system memory to the mem variable
mem_soft = (mem.total * .75) #the mem variable has total which outputs memory as integer, this assigns mem_soft a value of 75% of it
mem_hard = (mem.total * .90) #the mem variable has total which outputs memory as integer, this assigns mem_hard a value of 90% of it

num_format = "{:.0f}".format #this is to format the results into a number with no decimal places

#this is what combines the numbers retrieved from above and sets the full command as combined
setlimit = 'resource.setrlimit(resource.RLIMIT_AS, (' + num_format(mem_soft) + ',' + num_format(mem_hard) + '))'
exec(setlimit) #this executes the full command

#this is so that every system can log the current limits upon the start of the viewer
logging.info("The current resource limit is set as:")
logging.info(setlimit)
```

shaqaruden, 2021-02-23:
@ealmonte32 I have been updating viewer.py on all my pis. I have gotten to one where it breaks it. I get the following error in journalctl
```
Feb 22 20:41:38 RPI63 systemd[1]: Starting Screenly Viewer...
Feb 22 20:41:39 RPI63 xset[856]: server does not have extension for -dpms option
Feb 22 20:41:39 RPI63 systemd[1]: Started Screenly Viewer.
Feb 22 20:41:40 RPI63 python[858]: Traceback (most recent call last):
Feb 22 20:41:40 RPI63 python[858]:   File "/home/pi/screenly/viewer.py", line 52, in <module>
Feb 22 20:41:40 RPI63 python[858]:     exec(setlimit) #this executes the full command
Feb 22 20:41:40 RPI63 python[858]:   File "<string>", line 1, in <module>
Feb 22 20:41:40 RPI63 python[858]: ValueError: not allowed to raise maximum limit
```

Here is the code I added
```python
#memory consumption fix for viewer.py

#import the necessary modules
import resource
from psutil import virtual_memory

#this would be the base limits of ~3600MB and 3700MB as (soft,hard) limits, you can play around with these values to fit your needs
resource.setrlimit(resource.RLIMIT_AS, (600000000,700000000))

#this is the code that gets it done

mem = virtual_memory() #assigns all the results of the current system memory to the mem variable
mem_soft = (mem.total * .75) #the mem variable has total which outputs memory as integer, this assigns mem_soft a value of 75% of it
mem_hard = (mem.total * .90) #the mem variable has total which outputs memory as integer, this assigns mem_hard a value of 90% of it

num_format = "{:.0f}".format #this is to format the results into a number with no decimal places

#this is what combines the numbers retrieved from above and sets the full command as combined
setlimit = 'resource.setrlimit(resource.RLIMIT_AS, (' + num_format(mem_soft) + ',' + num_format(mem_hard) + '))'
exec(setlimit) #this executes the full command

#this is so that every system can log the current limits upon the start of the viewer
logging.info("The current resource limit is set as:")
logging.info(setlimit)
```

ealmonte32, 2021-02-23:
@shaqaruden 
I think that means the limits are too high for the specific hardware, can you send a quick screenshot or picture of the contents of `htop`?

shaqaruden, 2021-02-23:
After a refresh reboot. I thought it was odd as all my Pis are the same configurations using the same default image.

[image]

shaqaruden, 2021-02-23:
Here is the htop from another PI that I also added the same code to and has no issues.

[image]

ealmonte32, 2021-02-23:
Hmm.. weird, lets do an actual check on the command line, we'll go into python and execute the commands from there to see what the system sees itself as resource limits:

```
pi@rpi3blk:~ $ python2
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from psutil import virtual_memory
>>> mem = virtual_memory()
>>> print(mem)
svmem(total=844382208, available=470368256, percent=44.3, used=278007808, free=271962112, active=124903424, inactive=384131072, buffers=45232128, cached=249180160, shared=43089920, slab=44515328)
>>> 
```

Show me what the Pi with the issue outputs.

shaqaruden, 2021-02-23:
output of `cat /proc/meminfo`

```
pi@RPI63:~ $ cat /proc/meminfo
MemTotal:         875716 kB
MemFree:          302716 kB
MemAvailable:     417632 kB
Buffers:           29920 kB
Cached:           178620 kB
SwapCached:            0 kB
Active:           390336 kB
Inactive:          98808 kB
Active(anon):     299752 kB
Inactive(anon):    27504 kB
Active(file):      90584 kB
Inactive(file):    71304 kB
Unevictable:       18568 kB
Mlocked:              16 kB
HighTotal:        184320 kB
HighFree:           7384 kB
LowTotal:         691396 kB
LowFree:          295332 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                36 kB
Writeback:             0 kB
AnonPages:        299332 kB
Mapped:           127856 kB
Shmem:             46644 kB
KReclaimable:      13304 kB
Slab:              28904 kB
SReclaimable:      13304 kB
SUnreclaim:        15600 kB
KernelStack:        1416 kB
PageTables:         3948 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      437856 kB
Committed_AS:     929320 kB
VmallocTotal:     245760 kB
VmallocUsed:        4740 kB
VmallocChunk:          0 kB
Percpu:              464 kB
CmaTotal:         262144 kB
CmaFree:          243352 kB
```

output from python
```
svmem(total=896733184, available=418529280, percent=53.3, used=367972352, free=300519424, active=410009600, inactive=100372480, buffers=30720000, cached=197521408, shared=48500736, slab=29528064)
```

ealmonte32, 2021-02-23:
Ok I made the changes to mine as well and got the same error, so we simply need to compare the dependencies..

Show me `pip freeze` from one of the Pis that it works, lets compare to the one that doesnt.

ealmonte32, 2021-02-23:
ugh.. I think something miscommunicated somewhere, I think it's my fault for sending you the viewer.py memory code, I had written it as notes form, not the actual way it was fully written inside the viewer.py, basically you need to remove the first line that tries to set the limit:

```
# setting memory resource limits to prevent viewer from rendering the pi unresponsive due to high memory consumption
mem = virtual_memory()
mem_soft = (mem.total * .80)
mem_hard = (mem.total * .90)
num_format = "{:.0f}".format

setlimit = 'resource.setrlimit(resource.RLIMIT_AS, (' + num_format(mem_soft) + ',' + num_format(mem_hard) + '))'
exec(setlimit)

logging.info("The current memory resource limit is set as:")
logging.info(setlimit)
# setting memory resource limits finalized
```

This is the actual code I had on my old repo file..

shaqaruden, 2021-02-24:
I am testing this on one Pi for now. I set my soft limit to 0.50 and left the hard limit @ 0.90. I have applied the code and restarted and so fa so good.

Thanks @ealmonte32

ealmonte32, 2021-02-24:
@shaqaruden 
The final code I wrote on the post above should be 100% good to go, you should not need to lower the soft limit to 50%, you can put it back to 75%, 50% might be too low.

shaqaruden, 2021-03-09:
I'm going to call this solved

closed 2021-03-09 by shaqaruden