Screenly/Anthias #1497: latest build failed on production or master branch

Gifford47, 2021-02-16:
when i am trying to install screenly-ose on a raspi3 via "$ bash <(curl -sL https://www.screenly.io/install-ose.sh)" i am getting the following errors and screenly doesnt start:

on production (option1):
`fatal: [localhost]: FAILED! => {"changed": true, "cmd": ["python", "/home/pi/screenly/bin/migrate.py"], "delta": "0:00:00.163083", "end": "2021-02-16 09:57:14.656036", "msg": "non-zero return code", "rc": 1, "start": "2021-02-16 09:57:14.492953", "stderr": "Traceback (most recent call last):\n  File \"/home/pi/screenly/bin/migrate.py\", line 171, in <module>\n    migrate_add_is_enabled_and_nocache()\n  File \"/home/pi/screenly/bin/migrate.py\", line 139, in migrate_add_is_enabled_and_nocache\n    cursor.executescript(query_add_is_enabled_and_nocache)\nsqlite3.OperationalError: no such table: assets", "stderr_lines": ["Traceback (most recent call last):", "  File \"/home/pi/screenly/bin/migrate.py\", line 171, in <module>", "    migrate_add_is_enabled_and_nocache()", "  File \"/home/pi/screenly/bin/migrate.py\", line 139, in migrate_add_is_enabled_and_nocache", "    cursor.executescript(query_add_is_enabled_and_nocache)", "sqlite3.OperationalError: no such table: assets"], "stdout": "Obsolete column (filename) is not present", "stdout_lines": ["Obsolete column (filename) is not present"]}

PLAY RECAP *****************************************************************************************************************************************************
localhost                  : ok=38   changed=17   unreachable=0    failed=1    skipped=5    rescued=0    ignored=0

`

on master (option2):
`viewer.py:
ImportError: No module named ssl_match_hostname`

i tryied to install all requirements manually via "sudo pip install -r /home/pi/screenly/requirements/requirements.txt" but without success

vpetersson, 2021-02-16:
@Gifford47 I'd suggest that you switch to the development version as this is more up to date and will become the production version in the near future.

Gifford47, 2021-02-16:
ok, but i have already choosen the dev version in the install script. same result ...

vpetersson, 2021-02-16:
Strange.

Try this

```
cd ~/screenly
git pull
git checkout master
./bin/run_install.sh
```

Gifford47, 2021-02-16:
this is also not working:

TASK [system : Install build-essential] ************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to update apt cache: "}

PLAY RECAP *****************************************************************************************************************************************************
localhost                  : ok=21   changed=9    unreachable=0    failed=1    skipped=6    rescued=0    ignored=0

vpetersson, 2021-02-16:
Based on that error, you either got a corrupt disk image/SD card, or the connectivity is acting up. I'd recommend starting afresh with a new SD card and the latest Rasbian Lite image.

Gifford47, 2021-02-16:
my internet connection is ok and my sd card is also new.
i am also on a new buster image. but i am trying now the image "https://github.com/Screenly/screenly-ose/releases/tag/v0.18.3"

ealmonte32, 2021-02-18:
@Gifford47 
This reminds me of when I experienced that ssl match hostname error: https://github.com/Screenly/screenly-ose/issues/1427#issuecomment-748751862

I suggest for it to 100% work, please grab yourself a fresh copy of the Raspbian Lite OS directly from raspberry:
http://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-01-12/

Select the one that ends with `.zip`

Burn that with Etcher or whatever program you use onto your SD card.

Then install the Screenly master/dev branch through the bash script:
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`

This should work 99.99% of times (unless it is a physical SD card issue).

Gifford47, 2021-03-23:
still not working. i always get:

`pi@infoscreen:~/screenly $ ./viewer.py
Loading browser...
Viewer crashed.
Traceback (most recent call last):
  File "./viewer.py", line 464, in <module>
    main()
  File "./viewer.py", line 429, in main
    setup()
  File "./viewer.py", line 359, in setup
    load_browser()
  File "./viewer.py", line 257, in load_browser
    browser = sh.Command('ScreenlyWebview')(_bg=True, _err_to_out=True)
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 508, in __init__
    if not path: raise CommandNotFound(path)
CommandNotFound: None
Traceback (most recent call last):
  File "./viewer.py", line 464, in <module>
    main()
  File "./viewer.py", line 429, in main
    setup()
  File "./viewer.py", line 359, in setup
    load_browser()
  File "./viewer.py", line 257, in load_browser
    browser = sh.Command('ScreenlyWebview')(_bg=True, _err_to_out=True)
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 508, in __init__
    if not path: raise CommandNotFound(path)
sh.CommandNotFound: None
`

ealmonte32, 2021-03-24:
why are you running the viewer file by itself?

Gifford47, 2021-03-24:
to test why the screen does not start. if a reboot my pi, if i reboot my pi, in only get the splashscreen.... did anyone tryied the last release "QT, QTWebEngine and WebView v0.2" ?

Gifford47, 2021-04-05:
ok, with the last buster raspbian image works, thanks!

closed 2021-04-05 by Gifford47