Screenly/Anthias #1502: Screenly-Ose running on Balena - after upgrade asset not showing

jtinfotv, 2021-03-02:
Hi

I have run Screenly Ose on Balena succesfully earlier. Now I upgraded to release 61cb3e6. Upgrading went well and all assets ( web pages, jpg images, videos and youtube videos) are running correctly but one asset ( https://yle.idid.fi ) is showing white page and log shows this image 
[image]
 unable to get git branch and latest version from github

Then I flashed only screenly ose to memorycard and that page is working correctly on that but not with balena.

I have new class 10 fast memorycard…

So any help where might be the problem in Balena

#### Environment

* Raspberry Pi Hardware Version: Pi3+
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: N/A

shawaj, 2021-03-05:
Just linking to this same issue on the balena forum for reference https://forums.balena.io/t/screenly-ose-running-on-balena-after-upgrade-asset-not-showing/264521

M-MohitSharma, 2021-04-09:
@jtinfotv  can you please mention the Balena OS version that you are using. I am facing the same issue and videos are not being played on the screenly. 
I'm using Pi 3b+
balena OS 2.65 
sd card class 10 32gb

M-MohitSharma, 2021-04-09:
hi @shawaj  is there any workaround regarding the issue?

vpetersson, 2021-04-09:
@M-MohitSharma Based on your logs, you did not follow [the instructions for deploying to Balena](https://github.com/Screenly/screenly-ose/blob/master/docs/balena.md).

M-MohitSharma, 2021-04-09:
@vpetersson  thanks for the reply 
I tried depploying through `balena push` and `git push balena master` both were running .
but when I tried `./bin/deploy_to_balena.sh`  
it's giving me error 
[image]

vpetersson, 2021-04-09:
Looks like you might need Docker Engine installed.

M-MohitSharma, 2021-04-09:
@vpetersson after the installation of docker engine it was successfully built and pushed.
but still I'm not able to play video files on screenly.
it is showing in the logs that current URL is null and on display screen is black. 
please suggest a way around
[image]

vpetersson, 2021-04-09:
Not sure. Maybe check and make sure the GPU values are set properly.

M-MohitSharma, 2021-04-09:
I checked those. they are at 192 and videos are also encoded in h.264

M-MohitSharma, 2021-04-09:
can you suggest the balena OS version to use with rasp pi 3b+ 
currently I'm using 2.65 64bit

vpetersson, 2021-04-09:
I haven't tested this myself on Balena for a while. Maybe try the 32bit version. Not sure. The video playback process is very straight forward. It's just really calling on `omxplayer`, so that's what's failing for whatever reason.

M-MohitSharma, 2021-04-09:
[image]
I think something is wrong with this warning can this warning be avoided. 
I am trying 32bit balena OS but it is not booting up on my Pi. 
looks like I need to open an issue on balena forum.
Thanks for your help @vpetersson

closed 2022-01-09 by stale[bot]