Screenly/Anthias #1506: RPI4, This site can't be reached

Robiv8, 2021-03-10:
#### Environment

* Raspberry Pi Hardware Version: raspi4 b 8GB
* Raspberry Pi Network Setup: built-in WiFi, Ethernet
* Screenly OSE Version: latest over bash <(curl -sL https://www.screenly.io/install-ose.sh)

Raspi4 wit the latest Buster
all updates and upgrades are doing
installing screenly OSE developer version over the terminal
manage network = no
full system upgrade = no
installation without error. OK 

and?
no website no screenly
This site can't be reached
srly-ose-nginx's server IP dress could not be found

ERR_Name_not_resolved 

what can I do?

shaqaruden, 2021-03-18:
We are looking into this over here https://github.com/Screenly/screenly-ose/issues/1508

shaqaruden, 2021-03-18:
if you ran `docker ps` you will see that the nginx container is constantly restarting.

Try this

```
$ cd ~/screenly
$ git checkout master
$ git pull
$ ./bin/build_containers.sh
$ ./bin/upgrade_containers.sh
```

shaqaruden, 2021-04-21:
Please look at the last comment here https://github.com/Screenly/screenly-ose/issues/1508 to find the solution to this problem

closed 2022-01-09 by stale[bot]