Screenly/Anthias #1508: nginx container in restarting state

shaqaruden, 2021-03-15:
Fresh install from developmental branch on a RPI4. White screen with name resolution error for the nginx server.

running `docker ps -a` shows the container restarting.

running `docker logs screenly_srly-ose-nginx_1` shows the following

> nginx: [emerg] duplicate default server for 0.0.0.0:80 in /etc/nginx/sites-enabled/default:22
>  nginx: [emerg] duplicate default server for 0.0.0.0:80 in /etc/nginx/sites-enabled/default:22
>  nginx: [emerg] duplicate default server for 0.0.0.0:80 in /etc/nginx/sites-enabled/default:22
>  nginx: [emerg] duplicate default server for 0.0.0.0:80 in /etc/nginx/sites-enabled/default:22
>  nginx: [emerg] duplicate default server for 0.0.0.0:80 in /etc/nginx/sites-enabled/default:22

shaqaruden, 2021-03-15:
I tried to rebuild the image myself but I get an error saying "cannot connect to docker daemon"

vpetersson, 2021-03-15:
You might be running into [this bug](https://github.com/containerd/containerd/issues/5175) in Containerd. I ran into it last week when I tried building a new image.

shaqaruden, 2021-03-15:
I just tried running `docker run hello-world` and that worked fine. Do you guess a non standard docker host?

vpetersson, 2021-03-15:
Hmm. If you're able to do `docker run [...]`, you should be able to do `docker build [...]`. Try `sudo ./bin/build_containers.sh`.

shaqaruden, 2021-03-15:
I was trying to use docker-compose to build the image. switching to just docker build worked but when I try to start the container I get 

```
pi@raspberrypi:~/screenly $ docker-compose up -d srly-ose-nginx
WARNING: The DOCKER_TAG variable is not set. Defaulting to a blank string.
WARNING: The DEVICE_TYPE variable is not set. Defaulting to a blank string.
WARNING: The VIEWER_MEMORY_LIMIT_KB variable is not set. Defaulting to a blank string.
WARNING: The MY_IP variable is not set. Defaulting to a blank string.
ERROR: no such image: screenly/srly-ose-server:: invalid reference format
```

shaqaruden, 2021-03-15:
running the container yields a different error
```
pi@raspberrypi:~/screenly $ docker run screenly/srly-ose-nginx
nginx: [emerg] host not found in upstream "srly-ose-server:8080" in /etc/nginx/sites-enabled/srly-ose.conf:4
```

that container is running

vpetersson, 2021-03-16:
@shaqaruden so you can't run it with just regular `docker-compose`. There are some variables that needs to be set. After you've run `git pull` and `bin/build_containers.sh`, you need to run `bin/upgrade_containers.sh` That will automatically set all necessary variables.

bjornhij, 2021-03-16:
Same problem here:

nginx: [emerg] duplicate default server for 0.0.0.0:80 in /etc/nginx/sites-enabled/default:22
nginx: [emerg] duplicate default server for 0.0.0.0:80 in /etc/nginx/sites-enabled/default:22
nginx: [emerg] duplicate default server for 0.0.0.0:80 in /etc/nginx/sites-enabled/default:22

Container keeps restarting.

Maybe related to https://github.com/Screenly/screenly-ose/issues/1504 ?

vpetersson, 2021-03-16:
@bjornhij this is most likely due to the issue that was solved in a recent PR. Unfortunately my build Pi is broken at the moment due to [this bug](https://github.com/containerd/containerd/issues/5175), but I will try to build a new one soon.

In the meantime, try this:

```
$ cd ~/screenly
$ git checkout master
$ git pull
$ ./bin/build_containers.sh
$ ./bin/upgrade_containers.sh
```

That should hopefully do the trick.

shaqaruden, 2021-03-16:
I will try this

shaqaruden, 2021-03-16:
build_containers fails with

```
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-kErDwP/gevent/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-gfuE1X/install-record.txt --single-version-externally-managed --compile" failed with error code -9 in /tmp/pip-install-kErDwP/gevent/
```

shaqaruden, 2021-03-16:
Look through the output I came across this

```
byte-compiling /usr/local/lib/python2.7/dist-packages/gevent/_socket3.py to _socket3.pyc
      File "/usr/local/lib/python2.7/dist-packages/gevent/_socket3.py", line 227
        def makefile(self, mode="r", buffering=None, *,
                                                      ^
    SyntaxError: invalid syntax
```

shaqaruden, 2021-03-16:
I sent the output to a file
[output.txt](https://github.com/Screenly/screenly-ose/files/6152816/output.txt)

vpetersson, 2021-03-17:
Looks like `gevent` fails to build for whatever reason.

vpetersson, 2021-03-17:
Strange, I was able to build it fine myself (outside of Docker). Could you try adding `--no-cache` [here](https://github.com/Screenly/screenly-ose/blob/master/bin/build_containers.sh#L38)?

shaqaruden, 2021-03-17:
I modified as asked
```
docker "${DOCKER_BUILD_ARGS[@]}" \
        --build-arg "GIT_HASH=$GIT_HASH" \
        --build-arg "GIT_SHORT_HASH=$GIT_SHORT_HASH" \       
        --build-arg "GIT_BRANCH=$GIT_BRANCH" \
        --no-cache \
        -f "docker/Dockerfile.$container" \
        -t "screenly/srly-ose-$container:$DOCKER_TAG" .
```

but still got the error

```
byte-compiling /usr/local/lib/python2.7/dist-packages/gevent/_socket3.py to _socket3.pyc
      File "/usr/local/lib/python2.7/dist-packages/gevent/_socket3.py", line 227
        def makefile(self, mode="r", buffering=None, *,
                                                      ^
    SyntaxError: invalid syntax
[output.txt](https://github.com/Screenly/screenly-ose/files/6158137/output.txt)

...
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-GlynSq/gevent/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-bW7DwQ/install-record.txt --single-version-externally-managed --compile" failed with error code -9 
in /tmp/pip-install-GlynSq/gevent/
The command '/bin/sh -c pip install --no-cache-dir -r /tmp/requirements.txt' returned a non-zero code: 1

Liquidtouch, 2021-03-17:
> @bjornhij this is most likely due to the issue that was solved in a recent PR. Unfortunately my build Pi is broken at the moment due to [this [...]

I am having this same issue on a Raspberry Pi 4. I tried the above suggestion, in my case it all completes successfully but doesn't resolve the nginx duplicate server error.

shaqaruden, 2021-03-17:
@vpetersson may this have something to do with this issue with gevent?

https://github.com/gevent/gevent/issues/601

shaqaruden, 2021-03-17:
@Liquidtouch Could you run build_containers.sh again and output to a text file and upload it here?

```
$ cd ~/screenly
$ git checkout master
$ git pull
$ ./bin/build_containers.sh | tee output.txt
```

Liquidtouch, 2021-03-17:
Here's the output:

`Running without cross-compile...
buster: Pulling from balenalib/rpi-raspbian
Digest: sha256:9a8b27a17b39c4e5c3292813657866a11529c35e02b4714706400544c7d548b1
Status: Image is up to date for balenalib/rpi-raspbian:buster
docker.io/balenalib/rpi-raspbian:buster
Building base
Sending build context to Docker daemon  29.41MB

Step 1/3 : FROM balenalib/rpi-raspbian:buster
 ---> ec348eb482eb
Step 2/3 : RUN apt-get update &&     apt-get -y install --no-install-recommends         build-essential         cec-utils         curl         ffmpeg         git         git-core         ifupdown         libcec-dev          libffi-dev         libraspberrypi0         libssl-dev         libzmq3-dev         lsb-release         mplayer         net-tools         omxplayer         procps         psmisc         python-dev         python-gobject         python-pil         python-pip         python-setuptools         python-simplejson         sqlite3 &&     apt-get clean
 ---> Using cache
 ---> 70e2c4bf0f9c
Step 3/3 : RUN c_rehash
 ---> Using cache
 ---> 686bcfe5e6fc
[Warning] One or more build-args [GIT_BRANCH GIT_HASH GIT_SHORT_HASH] were not consumed
Successfully built 686bcfe5e6fc
Successfully tagged screenly/srly-ose-base:latest
Building server
Sending build context to Docker daemon  29.41MB

Step 1/13 : FROM screenly/srly-ose-base
 ---> 686bcfe5e6fc
Step 2/13 : ADD requirements/requirements.txt /tmp/requirements.txt
 ---> Using cache
 ---> 8ffdcf940120
Step 3/13 : RUN pip install --no-cache-dir -r /tmp/requirements.txt
 ---> Using cache
 ---> 650e117be98f
Step 4/13 : RUN mkdir -p /usr/src/app
 ---> Using cache
 ---> cc1d870c2450
Step 5/13 : COPY . /usr/src/app/
 ---> 5c8e5da884f2
Step 6/13 : WORKDIR /usr/src/app
 ---> Running in b1e54a2612f2
Removing intermediate container b1e54a2612f2
 ---> c87ce6022be5
Step 7/13 : ARG GIT_HASH
 ---> Running in df5c63db1c9f
Removing intermediate container df5c63db1c9f
 ---> 9a306fa8fcd2
Step 8/13 : ENV GIT_HASH=$GIT_HASH
 ---> Running in 7b32125203ae
Removing intermediate container 7b32125203ae
 ---> 8f21b26b02fa
Step 9/13 : ARG GIT_SHORT_HASH
 ---> Running in 85fd33911727
Removing intermediate container 85fd33911727
 ---> 6e5e8998bc99
Step 10/13 : ENV GIT_SHORT_HASH=$GIT_SHORT_HASH
 ---> Running in df1024b125e1
Removing intermediate container df1024b125e1
 ---> 69c6d055b4b1
Step 11/13 : ARG GIT_BRANCH
 ---> Running in ef3e121b299f
Removing intermediate container ef3e121b299f
 ---> c489c38ee37b
Step 12/13 : ENV GIT_BRANCH=$GIT_BRANCH
 ---> Running in 0ac325950681
Removing intermediate container 0ac325950681
 ---> 38c921b4fcd5
Step 13/13 : CMD ["bash", "bin/start_server.sh"]
 ---> Running in a5ab6223548b
Removing intermediate container a5ab6223548b
 ---> a5d8e93bc0e6
Successfully built a5d8e93bc0e6
[... 489 lines cut ...]
 ---> Using cache
 ---> e5e5a204e616
Step 21/27 : ARG GIT_BRANCH
 ---> Using cache
 ---> 8b4d2ef5c00d
Step 22/27 : ENV GIT_BRANCH=$GIT_BRANCH
 ---> Using cache
 ---> 29a6659bca50
Step 23/27 : RUN useradd -g video viewer
 ---> Using cache
 ---> 64ae789fa181
Step 24/27 : WORKDIR /usr/src/app
 ---> Using cache
 ---> 9aa533d5766f
Step 25/27 : RUN mkdir -p /usr/src/app
 ---> Using cache
 ---> 50ee4b341dfb
Step 26/27 : COPY . /usr/src/app/
 ---> fa7d60ca2737
Step 27/27 : CMD ["bash", "./bin/start_viewer.sh"]
 ---> Running in 45f234302343
Removing intermediate container 45f234302343
 ---> a13f3545bb76
Successfully built a13f3545bb76
Successfully tagged screenly/srly-ose-viewer:latest-pi1

shaqaruden, 2021-03-18:
@Liquidtouch comparing the output from you to mine shows that yours is using caches. Modify this file like @vpetersson https://github.com/Screenly/screenly-ose/issues/1508#issuecomment-801097237 and try again. Attach the output as well

Liquidtouch, 2021-03-18:
Here it is:

[output.txt](https://github.com/Screenly/screenly-ose/files/6166478/output.txt)

shaqaruden, 2021-03-18:
asking in the gevent repo I am being told pip is running out of RAM. I am going to reattempt this with a 2GB RPI4.

shaqaruden, 2021-03-19:
@vpetersson Good news and bad news, apparently the issue is not of memory. I was trying this on a 1GB RPI4 and it fails on a 2GB RPI4 is works fine. But running
```
$ bash <(curl -sL https://www.screenly.io/install-ose.sh)
```
 and selecting the master branch did not pull in the PR it seems. Running
 
 ```
$ cd ~/screenly
$ git checkout master
$ git pull
$ ./bin/build_containers.sh
$ ./bin/upgrade_containers.sh
```

I still have the same issue with the nginx container

vpetersson, 2021-03-19:
@shaqaruden interesting. Good find. So this should only prevent the actual build though I take it.

shaqaruden, 2021-03-19:
@vpetersson yes but in my case the rebuild still doesn't fix the problem

shaqaruden, 2021-03-19:
@vpetersson ok so I was able to get it working. I needed to comment out the pull command from upgrade_containers.sh. 

```
#sudo -E docker-compose \
#        -f /home/pi/screenly/docker-compose.yml \
#        -f /home/pi/screenly/docker-compose.override.yml \
#        pull

sudo -E docker-compose \
        -f /home/pi/screenly/docker-compose.yml \
        -f /home/pi/screenly/docker-compose.override.yml \
        up -d
```

vpetersson, 2021-03-22:
Yeah that makes sense @shaqaruden. The command you commented out will override your local build.

I'm still waiting for the issue in containerd to be fixed unfortunately.

Liquidtouch, 2021-03-25:
Awesome, this worked for me too! To clarify the steps I took so anyone else won't need to read through the whole thread and piece it together.

First modified the ~/screenly/bin/build_containers.sh file with:

> #sudo -E docker-compose \
> #        -f /home/pi/screenly/docker-compose.yml \
> #        -f /home/pi/screenly/docker-compose.override.yml \
> #        pull
> 
> sudo -E docker-compose \
>         -f /home/pi/screenly/docker-compose.yml \
>         -f /home/pi/screenly/docker-compose.override.yml \
>         up -d

Second modified the ~/screenly/bin/upgrade_containers.sh file with:

> docker "${DOCKER_BUILD_ARGS[@]}" \
>         --build-arg "GIT_HASH=$GIT_HASH" \
>         --build-arg "GIT_SHORT_HASH=$GIT_SHORT_HASH" \
>         --build-arg "GIT_BRANCH=$GIT_BRANCH" \ [...]

Third ran the following commands:

> $ cd ~/screenly
> $ git checkout master
> $ git pull
> $ ./bin/build_containers.sh
> $ ./bin/upgrade_containers.sh

SysLord, 2021-03-27:
@Liquidtouch 

Thanks for the TLDR. Just took way too long to realize that there are some spaces after one '\\' and thus the Dockerfile was not found yielding the confusing error: path " " not found.

> docker "${DOCKER_BUILD_ARGS[@]}" \$
>         --build-arg "GIT_HASH=$GIT_HASH" \$
>         --build-arg "GIT_SHORT_HASH=$GIT_SHORT_HASH" \       $    <----- here
>         --build-arg "GIT_BRANCH=$GIT_BRANCH" \$
>         --no-cache \$
>         -f "docker/Dockerfile.$container" \$
>         -t "screenly/srly-ose-$container:$DOCKER_TAG" .$

vpetersson, 2021-03-27:
Hi everyone. I pushed an updated image last night. Could you guys try just running through the regular upgrade process?

shaqaruden, 2021-03-27:
> Hi everyone. I pushed an updated image last night. Could you guys try just running through the regular upgrade process?

I won't be able to. I don't have access to any devices for at least a week

nkaminski, 2021-03-28:
> asking in the gevent repo I am being told pip is running out of RAM. I am going to reattempt this with a 2GB RPI4.

I have also ran into trouble when building on a Pi3 B+ with 1GB of RAM, I was able to build successfully by stopping all running screenly containers so as to free up more memory for the actual build process.

Liquidtouch, 2021-03-30:
Removed the extra spaces from my post which was all copy and pasted from above. I didn't encounter the issue because I was manually making the changes direct on my PI using Nano.

> @Liquidtouch
>
> Thanks for the TLDR. Just took way too long to realize that there are some spaces after one '\' and thus the Dockerfile was not found yielding the confusing error: path " " not found.

closed 2021-04-21 by shaqaruden