Screenly/Anthias #1511: Fresh install of the latest screenly-ose on Raspberry Pi 4B end with WSOD

Niubbo75, 2021-03-23:
#### Overview of the Issue

After a fresh install on a new Raspberry Pi 4B running Raspberry Pi OS Lite, using the installation script on github, choosing Develer version, I got a WSOD when rebooting. Also setting manually a WiFi connection doesn't fix the issue, I got a WSOD with error message:
This site can't be reached
srly-ose-nginx's server IP address could not be found.

Try:

- Checking the connection
- Checking the proxy, firewall, and DNS configuration

ERR_NAME_NOT_RESOLVED

#### Reproduction Steps

Steps to reproduce this issue:

* Create a USB stick with Raspberry Pi Imager and Raspberry Pi OS Lite
* Upgrade the OS to the latest
* Install Screenly-OSE with: $ bash <(curl -sL https://www.screenly.io/install-ose.sh)
* Choose Developer version
* Allow network management: Y
* Perform full system upgrade: N
* Reply to all questions until installation process ends
* Reboot raspberry

#### Environment

* Raspberry Pi Hardware Version: 4B 4 GB OS installed on 128GB USB3 Stick
* Raspberry Pi Network Setup: Built-in WiFi
* Screenly OSE Version: the one you get with $ bash <(curl -sL https://www.screenly.io/install-ose.sh)

When you start Raspiberry Pi OS it run w/o any issue, I have try also to reboot it and upgrade normally via apt, after Screenly-OSE installation I got WSOD, service srly-ose-nginx still restarting, I've managed to fix it but I can't connect in anyway to the net, both via WiFi or eth.
Any hint to fix the issue?
Thanks

vpetersson, 2021-03-23:
If you selected network manager, i'd try using `nmcli` or `nmtiu` to see if the network is correctly configured.

Niubbo75, 2021-03-23:
I find the issue right now, was an hardware problem, eth wall port was broken (or maybe it broke after reboot in screenly-mode).
BTW I have try to run #1508 and it seems to work but only with cable connection, don't work if I run it w/o a cable connected and I have no way to manage connections also if I had check to let screenly manage connections.
Now I'm re-install again on a fresh installation, let me see if now it will work after the fix or still have no wireless and no way to manage connections.

shaqaruden, 2021-03-24:
this should so the trick

‘’’
$ cd ~/screenly
$ git checkout master
$ git pull
$ ./bin/build_containers.sh
$ ./bin/upgrade_containers.sh
‘’’

Niubbo75, 2021-03-24:
Hello @shaqaruden and thanks for your reply.
I had give up and switched to balena screenly project that work fine out of the box, I just need to understand how to manage network connections via balena dashboard just because I need to install those four Pi on different places and network, I hope it will work, tomorrow I go to the office and try if it work from that network that it is different from the one in my house.
In case I will get other issues with it, I surely open a new ticket here! :-)

closed 2022-01-09 by stale[bot]