Screenly/Anthias #1516: self-signed certificates aren't accepted, touchscreen not working

Gifford47, 2021-04-05:
#### Overview of the Issue

I have add an asset like: https://192.168.1.100:8443/basicui/app?sitemap=main, but the viewer cant display it. It show the error message "ERR_CERT_AUTHORITY_INVALID", but i already set verify_ssl = False in the conf.
Also, the touchscreen/mouse no longer works with the latest Screenly version. i can't click anything.

#### Reproduction Steps

Add an asset with s self-signed certificate and show it on the screen.

#### Environment

* Raspberry Pi Hardware Version: RPI 3B+ with official 7" Touchscreen
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: WebView-v0.2, 19.12.2020

closed 2022-01-09 by stale[bot]

DarthDestroyer, 2023-06-23:
Found the same Problem!

Just wanted to use my local Page (only available with https) on the Screen, but "ERR_CERT_AUTHORITY_INVALID".

Wouldnt it be possible to add that Cert to the Viewer? Or something like that?
(verify_ssl isnt enought)

vpetersson, 2023-06-23:
Yeah exactly. The solution is to install the cert into the viewer container.

DarthDestroyer, 2023-06-23:
Is there a recommended way for that?

vpetersson, 2023-06-23:
No, just the regular way you would do it on Debian.

nicomiguelino, 2023-06-23:
Got it, @vpetersson

DarthDestroyer, 2023-08-07:
Found no way to get it working! moved the cert around, added it nearly everywhere but nothing worked as "expected"...

Any Info / ETA of that Documentation? Merci!

vpetersson, 2023-08-07:
Did you make the changes inside the container(s)? If not, it will not work. We'll add some tools for this later, but it's not something we have on the immediate roadmap.

DarthDestroyer, 2023-08-07:
Yeah, did it in the Container itself @screenly-anthias-viewer-1

vpetersson, 2023-08-07:
You also need to do it to the server container.

DarthDestroyer, 2023-08-07:
OK - did it on both containers - but no success!

Is it probably because of the trust store?
Like for example https://thomas-leister.de/en/how-to-import-ca-root-certificate/