Screenly/Anthias #1521: How can I add scrolling menu items?

HootisTigglebits, 2021-04-18:
I have made a menu board in photoshop and would like to add a gif to it.  I've noticed Screenly OSE displays a gif by itself, but when I add that same gif to my menu board in photoshop and export the entire image as a gif there is no animation.  What can I do to fix this?

* Raspberry Pi Hardware Version: 3B
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: production@d12df55

vpetersson, 2021-04-19:
Try switching to the developer version. It uses a newer browser that should hopefully support this (re-run the [installation](https://github.com/Screenly/screenly-ose#installing-on-raspbianraspberry-pi-os)).

closed 2021-04-19 by vpetersson

HootisTigglebits, 2021-04-19:
How do I switch to the developer version?

vpetersson, 2021-04-19:
You need to SSH into the device and then re-run the installation. If you don't have SSH enabled, you need to first [enable it](https://www.raspberrypi.org/documentation/remote-access/ssh/).