Screenly/Anthias #1522: Screenly OSE on Raspberry Pi 4 not working at all

fussel132, 2021-04-18:
Hey guys,
I stumbled over Screenly OSE a few days ago and thought this could be the best way to breathe life into an old TV that just takes room somewhere in the corner. I have a bit experience with RPis so I had some laying around and began to try installing Screenly. To increase the results performance, I firstly chose the RPi 4B, flashed Raspbian Buster Lite (directly from the [website](https://raspberrypi.org/downloads)) onto a 32GB SD card and inserted it into the pi. Connecting over SSH and ran the command to install Screenly (bash <(curl -sL https://www.screenly.io/install-ose.sh) and selecting 'production'). As far as I can say, it ran smoothly till it prompted to reboot. I said yes and after the reboot, I just saw the command line on screen. Running `docker pull hello-world` resulted in "Command not found: docker", so something definitely went wrong. I then also took the latest image from the releases here and tried again, after flashing and trying to boot the pi, nothing came on screen and the green operating lamp on the pi was constantly on for the next 30mins till I aborted it. I then took the RPi 3B and the latest release and e voila, everything works fine!
So my question is, would it be possible to somehow let it run on the 4B? Because especially when loading websites, the 3B is kinda slow and his memory gets pretty full after a short time. 

If you need anything like logs or so, please ask because to be honest, with docker and so I don't have any experience.

* Raspberry Pi Hardware Version: 3B and 4B
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: latest image (production@d12df55)

Thanks for your time,
Fussel 132

vpetersson, 2021-04-19:
The 'production' release is not compatible with the production release. You need to use the developer version for now.

closed 2021-04-19 by vpetersson

fussel132, 2021-04-19:
Um, where exactly do I choose the developer version or download it? I'm sorry, I can't find it at all...

vpetersson, 2021-04-19:
You need to follow [these](https://github.com/Screenly/screenly-ose#installing-on-raspbianraspberry-pi-os) instructions.

fussel132, 2021-04-19:
Okay sorry, didn't knew that master is the developer, but I'll try it later, thank you very much :)

arda-cli, 2021-04-19:
h

> Hey guys,
> I stumbled over Screenly OSE a few days ago and thought this could be the best way to breathe life into an old TV that just takes room somewhere in the corner. I have a bit experience with [...]

Have you tried turning it **off** and **on** again?

fussel132, 2021-04-19:
Wait I need a second to install it...