Screenly/Anthias #1525: Just installed the dev version and cannot pull up screenly ose gui.

HootisTigglebits, 2021-04-22:
I just finished installing the dev version of screenly ose and have it connected to my network wirelessly, but cannot pull up the Screenly OSE gui on my phone by going to its IP address. How can I fix this?

vpetersson, 2021-04-23:
Best guess is that either the installation failed or that the SD card has been busted (very common). If you're seeing anything like that, it likely didn't boot properly.

HootisTigglebits, 2021-04-23:
I’ve done the install twice now with no luck.   What do you mean “busted”?  I’ve never heard of that terminology when referring to a SD card before.

vpetersson, 2021-04-23:
Busted as in broken.

HootisTigglebits, 2021-04-23:
Thanks.  I reformatted the sd card and reinstalled the dev version.  I was able to get to the webpage gui, but have now ran into another issue that I just posted.

closed 2021-04-23 by HootisTigglebits