Screenly/Anthias #1526: Lags when loading a new image

HootisTigglebits, 2021-04-23:
I just got the dev version up and running and uploaded the same images I was using on the latest OSE version and have noticed that every time it goes to load a new image it loads the upper 1/8th of the image immedieatly and then lags for about 1 second and then loads the rest.  How can I fix this?

* Raspberry Pi Hardware Version: Model 3B
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: master@97a0abd

vpetersson, 2021-04-24:
Given your last issue, I wouldn't rule out a broken SD card. Try a brand new one and see if that resolves it.

usernsc, 2021-04-24:
I see the same at times on a pi 4b 8gb with 32gb sd depending on the image size. My thought was could be related to “conversion” time for the display.. but, I could be totally wrong here, as I am still “learning the nuances” of screenly-ose on the pi 4b..

HootisTigglebits, 2021-04-25:
Okay I bought a new SD card and am currently installing the dev version on it.  Fingers crossed.

HootisTigglebits, 2021-04-25:
Okay so it’s still lagging. Here’s a video of the issue. Before I upgraded to the dev version I was using these same images without an issue.

HootisTigglebits, 2021-04-25:
I think I may have figured it out.  The .png was using was a large file so I saved it as a .jpg and the loading lag seems to have stopped.  Now I just have to figure out how to adjust the image so that it fills the screen.  Is master@97a0abd the current dev version?

usernsc, 2021-04-25:
master@97a0abd is current, so lowering the image file-size - helped you? May I ask what the size and resolution was before and after you fixed it?

HootisTigglebits, 2021-04-25:
Yes but there’s a black screen between images that about .25 seconds which I’d like to get rid of. The original files were over 2mb and now they’re under 350kb.

HootisTigglebits, 2021-04-26:
I just put the regular OSE version on the SD card and it works flawlessly again other than the black bars.

ealmonte32, 2021-05-07:
I troubleshooted this with someone else [here](https://github.com/Screenly/screenly-ose/issues/1498) , this is simply the size and type of the file/image and how long it takes to render by the viewer.

closed 2022-01-09 by stale[bot]