Screenly/Anthias #1527: "Invalid -s parameter." when upgrading

brandonhygate, 2021-04-29:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

When upgrading Screenly via the web interface an error "Invalid -s parameter." is generated when the "Would you like to perform a full system upgrade as well?" toggle is activated.

#### Reproduction Steps

1. Visit the web interface
2. Click Settings
3. Under Upgrade Screenly, click the blue Upgrade button
4. Enable the "Would you like to perform a full system upgrade as well?" toggle
5. Click Start Upgrade

#### Environment

* Raspberry Pi Hardware Version: Model 3B Revision: 1.2 Ram: 1 GB Sony UK
* Raspberry Pi Network Setup: Built-In WiFi
* Screenly OSE Version: production@d12df55

ealmonte32, 2021-06-22:
Hmm.. I will try to test this simple issue, weird though because `-s` is a legit `read` command parameter:
https://ss64.com/bash/read.html

brandonhygate, 2022-01-10:
Tested and seems to be resolved for me!

Thanks yall

closed 2022-01-10 by brandonhygate