Screenly/Anthias #1530: Asset url behind a proxy gateway in school

Einstein1987, 2021-05-14:
Hello, I installed Screenly OSE on a raspberry pi 3 model B. Everything works fine. I then installed it in my school which has a proxy. Screenly works fine for images for example but for the url I get the error message due to the proxy. However, I have correctly configured the proxy in the raspberry conf files, and the chromium browser works correctly. Is there a specific configuration file for screenly so that I can indicate the IP of the gateway ?

Thank you.

vpetersson, 2021-05-14:
I don't think we've really tested anything with proxies. It's a fairly rare setup these days. It's probably possible to get it to work, but I don't think we can afford spending development cycles on this.

Einstein1987, 2021-05-14:
Thank you for your reply. I just understood that the browser used in screenly OSE is uzbl, so I will try to see if there is a config file to modify to take into account the proxy ip. Like chromium does on the raspberry pi.

vpetersson, 2021-05-17:
The "old" version is indeed using Uzbl (i.e. the one you get from the last disk image). You might have better luck using the development version installed on the latest Raspbian release.

Einstein1987, 2021-05-31:
Hello,
I did install OSE by first installing raspberry pi then entering the bash command. I have many other browsers (Chrome, Dillo) but ScreenlyOSE to only use uzbl. Is there a way to force screenly to use chromium that works fine behind the proxy. I thought I managed uzbl to get it to work but for a few days no page opens and I have an error message.
Thank you in advance for your help.

vpetersson, 2021-05-31:
> Is there a way to force screenly to use chromium that works fine behind the proxy

No, Chromium does not have an API that is compatible with the way Screenly OSE works.

Einstein1987, 2021-05-31:
So we are forced to use uzbl?  So I will continue to find out how to configure the proxy on uzbl ... :(

vpetersson, 2021-05-31:
The developer version is using a QT based browser. You can try that out and see if that works better for you.

Einstein1987, 2021-05-31:
Ah very good news and very good idea. How to switch to the development version?

vpetersson, 2021-06-01:
See https://github.com/Screenly/screenly-ose#installing-on-raspbianraspberry-pi-os

Einstein1987, 2021-06-24:
Hello,

I come back to you. I reinstalled everything from scratch using the downloaded full screenly-OSE image. Everything is installed correctly on the SD card of the raspberry PI. Once installed I was able to set a fixed ip address and entered the proxy of my school for the raspberry using "nmtui" then I entered the proxy in /etc/apt/apt.conf.d/proxy.conf for that apt is working. This seems to work for apt so it can run updates.

I then informed the proxy in / etc / environment by typing: export http_proxy = "http: // (ipproxy): (portproxy)
export https_proxy = hpps: // (ipproxy): (portproxy)
and I also populated the proxy in ./.config/uzbl/config-screenly by typing: set proxy_url = https: // (ipproxy): (portproxy)

I still get the following error message when viewing regardless of the asset. What is missing ?
[https://drive.google.com/file/d/1xk8qeYZNp1qwgeO0DMzmlqDi25vXGndk/view?usp=sharing](url)

Einstein1987, 2021-06-28:
Hello,
So this weekend I started all over again to try to get the development version and thus have the player other than uzbl. For that I put raspberry lite on the sd card then on the pi I launched the bash command (<curl -sL https://screenly.io/install-ose.sh). At home, without a proxy, everything works fine. This morning I am installing the raspberry pi at school with the proxy. I define a fixed ip address using the "nmtui" command and I also fill in the gateway. I also inform the proxy in raspi-config. I get the following error message: "Internal Server Error. The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

Is there another place where I can fill in the proxy?

Einstein1987, 2021-06-29:
Hello,

I think, based on my research, that you need to fill in the proxy as an environment variable in docker. I wanted to follow the doc: https://docs.docker.com/config/daemon/systemd/
But the environment variable does not seem to be taken.

I wanted to do method 1 from the doc: https://www.thegeekdiary.com/how-to-configure-docker-to-use-proxy/
But / etc / sysconfig / docker does not exist.

Do you know where I can give the proxy to docker?

hsilva090, 2021-07-06:
Hi all,
I'm after the same solution, and I think I found the problem.

After configure proxy in docker client and test it (everything OK), the *.github.com appears with incorrect SSL certificate (https://140.82.121.6)

Einstein1987, 2021-07-13:
Hello, 
how do you go about setting up the proxy in the docker client?  where are the configuration files located?  
I'll try to look this summer at how to get screenly dares to work behind the proxy.

closed 2022-04-16 by stale[bot]