Screenly/Anthias #1533: how to make changes in the templates

pittipuk, 2021-05-25:
Hello, I want to make a few changes in the html documents in /home/pi/screenly/templates.
But how does screenly show the changes after a reboot? 
Have the changed htmls to be loaded into the docker container or what has to be done?
Thanks for a little help and your nice work!
Greetings!

closed 2021-05-25 by vpetersson

pittipuk, 2021-05-25:
I  suspected something like this ;-) Thank you!
Is there something like an instruction how to fork and build ....?

vpetersson, 2021-05-25:
No, there aren't any formal instructions for this unfortunately. Reading the `docker-compose` files should give you a good start however.

pittipuk, 2021-05-25:
I tried something like:
	cd ~/screenly
	./bin/build_containers.sh
	./bin/upgrade_containers.sh
but without success.

vpetersson, 2021-05-25:
No, because that will pull images from upstream and overwrite your local containers. Read over the scripts and docker-compose file and it should become pretty clear what needs to be done.