Screenly/Anthias #1534: Screenly OSE working with folder files

JhonathanJJGS, 2021-05-27:
Hello, i'm trying to use screenly with one local server to provide the assests, i'm using the html css background image to provide the images, but i need to do the same with video, there is any work around? 

Scope: 
I want to configure screenly in one lan without internet access, and to make the things easy i cant use the web interface to provide the assests. 
Need to do something like place the file in one folder and automatic screenly starts to presents the content. 
As i said in the begin, i'm using the html way that takes the image as the body background, but i need to do the same with videos;

 
Screenly version: 
Last devolper compilation;

Hardware pi2b

vpetersson, 2021-05-27:
Long story short, don't use a Raspberry Pi without RTC if you don't have a local NTP server on the network.

With regards to your idea, you probably want to use the API for that.

JhonathanJJGS, 2021-06-07:
Thank you helped me alot, now thats working with api!!

closed 2021-06-07 by JhonathanJJGS