Screenly/Anthias #1537: Screenly portrait mode on PI4

JhonathanJJGS, 2021-06-07:
Hello! 

first i would like to thank everyone involved for this project! really really thank you all! 

I'm having trouble rendering videos in portrait mode on raspberry pi4 with the develop version of screenly ose.

Images are normally rendered centered and full screen (depending on resolution), but videos are rendered in the corner of the screen and end up with only half a sample.

any one knows the solution? 

Thank you!

JhonathanJJGS, 2021-06-09:
Seems to be something with  vlc beeing renderized in te first corner of the screen something like: 

Landscape:                                              Portrait: ____________
 ___________________________                                   |*                 |
|*                                       |                                   |                  |                      
|                                         |                                  |                   |
|                                         |                                  |                   |
-----------------------------                                   |____________|

Maybe vlc can't handle with the screen manager, any sugestions?

Thanks!

avoss19, 2021-06-17:
I ran into a similar issue with VLC running on a Raspberry Pi 3 on RaspberryOS, but then downgraded to Raspbian 9 which resolved the issue. This does not seem to be a Screenly problem.

avoss19, 2021-07-01:
I found a possible work around for the time being. Try editing /boot/config.txt and comment out dtoverlay=vc4-fkms-v3d. This disables the new graphic drivers on the Pi 4 and reverts to the legacy drivers

JhonathanJJGS, 2021-07-04:
> I found a possible work around for the time being. Try editing /boot/config.txt and comment out dtoverlay=vc4-fkms-v3d. This disables the new graphic drivers on the Pi 4 and reverts to the legacy drivers

Yes, with whit a can HDMI rotate and diplays in 180º but vlc cant handle with portratir, so the solution was to diasable vlc in python code and comeback with omxplayer

closed 2022-04-16 by stale[bot]

Ezana-AI, 2024-01-03:
Has anyone found a solution to rotate the screen on a Raspberry Pi? Can anyone help me?

cmdrtom, 2024-02-14:
I have updated a very old Screenly OSE to Anthias. After this the rotation is false. It is not possible to rotate the monitor with Anthias. With Screenly OSE on the Raspberry Pi it works fine. So I get back to my good old Screenly OSE solution where the rotation is right.

jdrosenboom, 2024-09-04:
In case anyone finds themselves on this thread in the future, below is what worked for me. These were all Pi 3Bs.

# Uncomment to disable DRM VC4 V3D driver
# dtoverlay=vc4-fkms-v3d

# 1 = Left, 3 = Right
display_hdmi_rotate=1