Screenly/Anthias #1538: how to get rid of the loading .png

Webghost2015, 2021-06-10:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

#### Environment

* Raspberry Pi Hardware Version:
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter)
* Screenly OSE Version:

Webghost2015, 2021-06-10:
Hello,

I'm new but like the program very much.
Inbetween movies there's a one second black. that's not the problem. But after a reboot it's a black with screenly logo inside it.
It comes from the loading.png picture. I want to shut that down.
Is that possible. tried a lot things reneming the static files but that doesn't work.

I have a pi4b 8 G 32 G card in it. I have one PI working but that was the installation in march. 
All the new ones don't work . 

I hope someone has an answer.

Webghost2015, 2021-06-10:
An other way to get rid of it is to disable all the assets en enable them again

vpetersson, 2021-06-11:
No this isn't possible in OSE. It would require a major rewrite of the entire system.

ealmonte32, 2022-03-02:
> I have a pi4b 8 G 32 G card in it. I have one PI working but that was the installation in march. All the new ones don't work .
> 
> I hope someone has an answer.

What do you mean all the new ones dont work? `raspbian buster lite armhf`  image and installation script dont work?
try this: https://github.com/Screenly/screenly-ose/issues/1571#issuecomment-1055781436

closed 2024-06-06 by nicomiguelino