Screenly/Anthias #1539: Unable to view Video in Dual screen

BrianPHC, 2021-06-14:
#### Overview of the Issue

We have just started using the Raspberry Pi4B hoping to be able to use it with two screens to display Screenly OSE content (we do have a number of old RPi3B that use single screen and work fine). Images and web content appear fine on both screens, However, when a video asset is displayed it only appear on one of the screens - the primary screen on the left. 

* Raspberry Pi Network Setup: 
Ethernet with PoE HAT

* Screenly OSE Version:
      Version downloaded and installed on Raspbian Lite on 11 June 2021

Cheers

JonnyTech, 2021-06-14:
This seems to be a limitation of the hardware, it can only decode a video stream to a single (primary) screen at any one time. There are various operating system drivers/tricks that try to overcome thus but I have not found anything reliable.

vpetersson, 2021-06-14:
I believe you can work around this by setting the screens to mirror. However, you will not be able to operate them independently.

closed 2022-04-16 by stale[bot]