Screenly/Anthias #1542: Feature Request: Enable mouse supper

Adrian-at-CrimsonAuzre, 2021-07-23:
We have a web app that we use in production in conjunction with touch screens and I would *love* to be able to use Screenly OSE to manage these. Would it be possible to add a flag to re-enable mouse support? A way to manually re-enable it would also work for the time being.

Adrian-at-CrimsonAuzre, 2021-07-23:
After a little more searching I found a comment on [#238](https://github.com/Screenly/screenly-ose/issues/238#issuecomment-314436449) which does work for me. Would be great to allow us to enable this through the API, add a big ass flashing warning message if security is the main concern.

closed 2022-04-16 by stale[bot]

ofifoto, 2022-04-16:
Might be a good idea to turn off this bot since the project is overall somewhat inactive