Screenly/Anthias #1545: would like to see a setting for using the screen in Portrait or Landscape mode

soundtechscott, 2021-07-29:
Feature Request
Since a lot of times signage can be in portrait mode, it would be nice to have a checkbox or a setting that can be changed to automatcally rotate all asset's

I use this for signage at home as Movie posters, and all my posters are in Portrait mode, so I have to manually roatate them before uploading them

If such a button already exists, please let me know

damienpeerbolte, 2021-11-22:
You can probably make a button on the settings page that executes a command or shell file that edits the config.txt with the `sed` command

qbicdesign, 2022-02-11:
This is simple enough to do in /boot/config.txt  by adding
display_rotate=3

soundtechscott, 2022-04-16:
This doesnt actually work....
it still assumes screen size is only the width, so it makes a tiny landscape screen in the middle of the portrait layout instead of using the entire portrait resolution
Meaning if you have 1920x1080 -  which becomes 1080x1920, it only make s 1080x608 screen in the middle of it when rotated to portrait mode

qbicdesign, 2022-10-11:
lt does work.
I'm using it in 3 installations currently.
If you're rotating the screen and uploading a portrait image, that's
effectively 2 rotations
Rotate all you portrait images -90 first and you're good to go