Screenly/Anthias #1546: Display a website with basic authentication

haupt-pascal, 2021-07-29:
#### Overview of the Issue

I'd like to display a private admin-dashboard, which uses basic authentication for the website (username and password ...). I know that there is a feature in the pro-version to add the data with an hash (or something like that), but is there a way to do this in the open source version as well? If not, is this an possible OSE feature? 

#### Environment

* Raspberry Pi Hardware Version:
* Model 3B Revision: 1.2 Ram: 1 GB Sony UK
* Screenly OSE Version: production@d12df55

kaveyro, 2021-07-30:
I'm not sure if this works with the Screenly Browser but you could try to pass the user credentials to the website like this: https://username:password@URL

haupt-pascal, 2021-08-03:
Forgot to answer - nope this doesn't work, I've already tried before this issue.

haupt-pascal, 2021-09-14:
Nobody else?

nkaminski, 2021-09-17:
@haupt-pascal One solution for websites requiring authentication whether it be basic auth, bearer tokens, or mTLS is to run an nginx container configured as a reverse proxy alongside Screenly that listens on localhost of the Pi. Have been using this myself for some time with reasonable success.

andyhovest, 2022-02-03:
There are many sites that this would open up for viewing if this were possible since people often save preferred or favorited configurations, such as weather sites, online calendars, and others.

ealmonte32, 2022-03-01:
Since this is the OSE edition, it would take contributors to write the code to make something like that happen since it is not easy and also some browsers dont even support this method anymore.

@haupt-pascal 
What kind of authentication does this private admin dashboard use?

jeremy-brennan, 2024-05-13:
I see this is a few years old now but did this ever get solved?

vpetersson, 2024-05-13:
This is not supported yet in Anthias (but is supported in Screenly).

djeepy971, 2025-11-13:
I wondering if 
 now it's possible ?