Screenly/Anthias #1551: Javascript Webpage won't display

wazzasay, 2021-08-25:
#### Overview of the Issue

I have a webpage notice board displayed via the SEQTR Education platform and is it not displaying on Screenly OSE. In my trial and error I have discovered it doesn't display on internet explorer 11 either displaying these current errors in consol. 

SCRIPT1002: Syntax error
lib.js (1,214291)
SCRIPT1002: Syntax error
app.js (1,7487)
SCRIPT5009: 'Locale' is undefined
notices.js (1,69)

Is this something that can adjust on the Pi to make it run this? As the notice page is quite key to the deployment. 
#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ... https://notices.ignatius.sa.edu.au/
* Press "On"
* Wait for the asset to show up on the screen

#### Environment

* Raspberry Pi Hardware Version: Pi3b+
* Raspberry Pi Network Setup: Ethernet 
* Screenly OSE Version: Production

wazzasay, 2021-08-25:
I Have found some more information that might be helpful. It looks like it may be a compatibility issue with ECMAScript 6 and 5. My guess is that ES5 is run on screenly and needs ES6?

vpetersson, 2021-08-25:
Try switching to the experimental branch.

wazzasay, 2021-08-25:
> Try switching to the experimental branch.

I have tried a couple of times even from a fresh boot but it doesn't actually display anything. Just crashes and stays on a blanks screen, any suggestions ?

ealmonte32, 2022-02-23:
@wazzasay Just tested this and website loaded perfectly fine for me, running development branch .. I can see the schedules and announcements on the site, etc..

closed 2025-03-06 by nicomiguelino