Screenly/Anthias #1552: Force periodic reloads of repeated web assets

LuxObscura, 2021-08-31:
#### Overview of the Issue

Hello, I have a problem with the display of my web page.
I have two assets that run at 10s intervals:
http://172.26.14.110:82/TableauMesures.htm?show=1
http://172.26.14.110:82/TableauMesures.htm?show=2

I should be able to update my content.
Refresh works at the start. But if I no longer update the content of my web page for a few hours then the update takes 30min or more for this update.

I would like to be able to force the update or delete the "cache".

How to do ?

Thanks for your help
#### Environment

* Raspberry Pi Hardware Version: Model 3B Revision: 1.2 Ram: 1 GB Sony UK
* Screenly OSE Version: production@d12df55

LuxObscura, 2021-08-31:
When I restart screenly-viewer.service the display shift works.
I tried to understand where the problem lies and I found this information:
File: /home/pi/screenly/viewer.py

line 456 # FIXME If we want to force periodic reloads of repeated web assets, force = True could be used here.
line 457 # See e38e6fef3a70906e7f8739294ffd523af6ce66be.

What is e38e6fef3a70906e7f8739294ffd523af6ce66be?
And i dont understand line 456, i just edit and write force = True?

Thanks for your help

dsuguy2007, 2024-05-11:
I'm having the same issue. The Web page I created has a clock on it so it needs to update on every display. It's not doing this.

ealmonte32, 2024-05-12:
> I'm having the same issue. The Web page I created has a clock on it so it needs to update on every display. It's not doing this.

I have two suggestions for you:

1. If you are the manager of the website you are loading, look into using the HTML [meta refresh](https://www.w3schools.com/tags/att_meta_http_equiv.asp) attribute.
2. Instead of only using one web asset, add a second simple web asset like the Anthias Clock widget or Weather one, and set it for 10 seconds or so, that way you will be able to "reload" the web asset you want without needing to edit any of the python code if its not something you are familiar with or easily done.

If none of these solve your problem or at least give you a workaround, and you don't have a way to control the clock on your website dynamically without changing the website or refreshing it, I can't think of something else that will work for you.

vpetersson, 2026-05-08:
Solved properly in https://github.com/Screenly/Anthias/pull/2841