Screenly/Anthias #1554: RPI4 issues with 4k display

madman2021, 2021-09-17:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue
RPI4 on a 4k display seems to have weird behaviour.

With dtoverlay=vc4-fkms-v3d 
Sometimes images display other times it shows just white, websites always flicker a tiny amount of the website and then goes white.

Without dtoverlay=vc4-fkms-v3d
Images and websites seem to fill the screen vertically but has black borders on the left and right

#### Reproduction Steps
RPI4 into 4k display showing any website.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 Model B Rev 1.2
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter): Tried with Ethernet and WiFi
* Screenly OSE Version:master@97a0abd

madman2021, 2021-09-17:
I enabled "WEBVIEW_DEBUG" by adding it to the docker compose file, in the viewer output logs I noticed this
qt.scenegraph.softwarecontext.renderer: render QRegion(0,0 1920x1280) 0 0 18
With `tvservice -s`
state 0xa [HDMI CEA (95) RGB lim 16:9], 3840x2160 @ 30.00Hz, progressive

madman2021, 2021-09-17:
Also in the viewer logs
[545:7:0917/113950.453756:FATAL:memory.cc(38)] Out of memory. size=262144

madman2021, 2021-09-18:
Seems this issue was caused because of the default memory limits of /dev/shm in docker  (64MB),  adding `shm_size: '1g'` to  the docker-compose file solved the issue.

I'm  going to do some  more research and proably present a pull  request if I don't find any negatives to altering it.

tyler351, 2021-09-22:
Just ran into this same issue. Newbie to docker.... Where/how would you go about making this adjustment?

madman2021, 2021-09-25:
@tyler351 In the docker compose file `docker-compose.yml` add a link under the viewer service `shm_size: '1gb'`
```
 srly-ose-viewer:
    shm_size: '1gb'
    image: screenly/srly-ose-viewer
    build:
      context: .
      dockerfile: docker/Dockerfile.viewer
    depends_on:
      - srly-ose-server
    environment:
      - HOME=/data
      - PORT=80
      - NOREFRESH=1
      - LISTEN=srly-ose-nginx
    privileged: true
    restart: always
    volumes:
      - resin-data:/data
      ```

tyler351, 2021-09-28:
> @tyler351 In the docker compose file `docker-compose.yml` add a link under the viewer service `shm_size: '1gb'`
>
> ```
>  srly-ose-viewer:
>     shm_size: '1gb'
>     image: screenly/srly-ose-viewer [...]

Ah. gotcha. Ok that makes sense. Thanks so much!

closed 2022-04-16 by stale[bot]