Screenly/Anthias #1556: Admin Page Going Down

techminer71, 2021-09-28:
#### Overview of the Issue

I'm in charge of the maintenance of around 10 Screenly screens, and periodically throughout the day, or week some of them will disconnect. as in, the admin site will go down, but it will continue to display everything correctly. I have searched around and could not find an answer or at least am unsure of what I'm searching for. 

#### Reproduction Steps
I cant really try to reproduce it because it will be working one minute then it wont the next
 

#### Environment

* Raspberry Pi Hardware Version: RbPi 3B+
* Raspberry Pi Network Setup: Built-in WIFI
* Screenly OSE Version:10.1

colegreat06, 2021-10-10:
+1 on this.. having the same problem.. I manage about 6 Screenly screens and there isn't really ever a time where the admin page is online on all of them at once. 

Also haven't found a way to reproduce, just randomly happens.

All but 2 are wireless. 4ish Pi 3b+'s and 3ish Pi 0 w's.

ealmonte32, 2022-02-23:
if you check the logs you would see that either the server process is dead, or the pi lost wifi.. journalctl and look for network disassociations.. or just any error in general.. post here, otherwise impossible to help troubleshoot.