Screenly/Anthias #1559: Can not install Experimental fork on RPi3/4 manually

wazzasay, 2021-11-01:
#### Overview of the Issue

I am trying to manually install the experimental branch of screenly-ose and it keeps getting stuck on the boot screen of screenly. Is anyone having any other luck?

#### Reproduction Steps

fresh image of RPi Lite and then ran the install script bash <(curl -sL https://www.screenly.io/install-ose.sh) and picked experimental. 

will install but then will not boot up fully.

#### Environment

* Raspberry Pi Hardware Version: RPi 3B+/4
* Raspberry Pi Network Setup: Ethernet

Samba89, 2021-11-09:
Raspberry Pi OS updated to Bullseye recently (30th Oct 21) For me, the Screenly installation script starting throwing out "package has no installation candidate" errors from apt. First guess is an issue with the screenlyapp repo. Bullseye also has Python 3.9 as default which I imagine will cause more issues.

Try Buster for now.
https://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/

werde45, 2021-11-10:
Hi. Installation seems to work with Buster but will not work. Any help would be good.

PhatBill52, 2022-02-01:
Hi

Same here :
As _2020-07-17-Screenly-OSE-lite_ doesn't seem to boot on my RPi 4 B 2GB, I tried _bash <(curl -sL https://www.screenly.io/install-ose.sh)_ after installing _2022-01-28-raspios-buster-armhf-lite_

Play recap seems ok, but all I get is a blank screen after the splash screen.
Ping and SSH ok, but HTTP access gives me Internal Server Error.

Any ideas please ?

modkin, 2022-02-17:
Hello,

is there any idea here? I am using a RPi 4 B as well. Some people seem to be lucky with getting that up.
After installing with `bash <(curl -sL https://www.screenly.io/install-ose.sh)` I just get a black screen.
I can however log in after switching with CTRL + ALT + 1
Not sure if it helps but when I check the docker logs with 
`docker logs screenly_srly-ose-viewer_1`
I see:
`
TypeError:  string indices must be integers, not str
`
coming from: `raspberry_pi_helper.lookup_raspberry_pi_revision(raspberry_pi_revision)['model']`

any hints would be helpfull.

vpetersson, 2022-02-17:
> Hello,
>
> is there any idea here? I am using a RPi 4 B as well. Some people seem to be lucky with getting that up. After installing with `bash <(curl -sL https://www.screenly.io/install-ose.sh)` I [...]

Can you post the output of `grep Revision /proc/cpuinfo`? It's likely a model that we haven't added in to the database yet.

modkin, 2022-02-17:
Thanks for the quick reply.
grep says:
`b03115`
it is a Model B with 2GB RAM

modkin, 2022-02-17:
It looks like this revision is indeed missing in `raspberry_pi_helper.py`
How do I add it properly? I just added it for testing and ran the install script but still the same error.
Do I need to update the docker images or something?

ealmonte32, 2022-02-23:
From basic troubleshooting, I think Bullseye breaks some things because of outdated/missing/replaced/incompatible packages/libs most likely, can you guys just try it with raspian os buster lite?
If the initial installation fails at any point (like for me a fresh install failed on ansible task to install docker-ce and this prevented the containers from being built), I simply ran `./screenly/bin/run_upgrade.sh` and tried again. This time the containers and all packages should not timeout and install.
Also, before any installation, run `sudo apt update -y` and accept the changes to oldstable. Also sudo apt full-upgrade -y might be a good idea as well.

ealmonte32, 2022-03-01:
> It looks like this revision is indeed missing in `raspberry_pi_helper.py` How do I add it properly? I just added it for testing and ran the install script but still the same error. Do I need to update the docker images or something?

I troubleshooted this with someone else on another post, https://github.com/Screenly/screenly-ose/issues/1571
Yes you need to rebuild the containers so that when they are created they contain the new revision board but before you do that I would stop all containers:
`docker container stop $(docker container ls -q)` 

then i would prune the system:
 `docker system prune -a`
so that it clears out all of the already created containers and build cache as well.

since we are pruning all containers we might not need the `--build` parameter in the upgrade_containers.sh script, but it's been a hit or miss for me when I try to rebuild containers after modifying files in the source folders.. you can try to rebuild without it first, then with it if the raspberry pi helper file didnt change on the container built.

modkin, 2022-03-02:
Nice, patching up the system worked. The only missing part was adding `--build`

Thanks for the help

ealmonte32, 2022-03-02:
> Nice, patching up the system worked. The only missing part was adding `--build`
> 
> Thanks for the help

You didn't have to change celery and urllib3 requirements to previous compatible versions?

modkin, 2022-03-03:
Yes you are right.
It was commit `e090ac8e0a674deb9383d92f6bc32fac5ecb764e` using a Buster Lite image.
I had to revert celery to 4.4.7 and of course add the `b03115` revision into `raspberry_pi_helper.py`

vpetersson, 2022-09-23:
Closing this out as it should now work.

closed 2022-09-23 by vpetersson