Screenly/Anthias #1560: Whats holding up a new sprint release

shaqaruden, 2021-11-03:
Looking at the project for the next sprint everything is completed. If you communicate what still needs to be done maybe I can help out in some way.

MoJo2600, 2021-11-26:
Just found this software for my screen and would like to use the next version, too

schmaddi, 2022-02-02:
here too

shaqaruden, 2022-02-14:
@MoJo2600 @schmaddi for now if you want the new release, install Rapsberry Pi OS and install screenly using the curl command in README.md

MoJo2600, 2022-02-14:
Thank you! First thing tomorrow!

MoJo2600, 2022-02-15:
The install script does not work on the current release of Raspberry Pi OS lite. It seems to depend on python2 as it tries to install `python-pip`. So i installed `python3-pip` manually, but the result was an Ansible error.

What I have done:

```
# On a fresh install of Raspberry Pi OS Lite
$ sudo apt install python3-pip
$ bash <(curl -sL https://www.screenly.io/install-ose.sh)
```

The error I received was on dev branch:

```
+ sudo -E ansible-playbook site.yml --skip-tags system-upgrade
ERROR! the playbook: site.yml could not be found
```

on master branch:

```
+ sudo -u pi ansible localhost -m git -a 'repo=https://github.com/screenly/screenly-ose.git dest=/home/pi/screenly version=production force=no'
ERROR! Unexpected Exception, this is probably a bug: cannot import name 'AnsibleCollectionLoader' from 'ansible.utils.collection_loader' (/usr/local/lib/python3.9/dist-packages/ansible/utils/collection_loader/__init__.py)
the full traceback was:

Traceback (most recent call last):
  File "/usr/local/bin/ansible", line 90, in <module>
    mycli = getattr(__import__("ansible.cli.%s" % sub, fromlist=[myclass]), myclass)
  File "/usr/local/lib/python3.9/dist-packages/ansible/cli/__init__.py", line 22, in <module>
    from ansible.inventory.manager import InventoryManager
  File "/usr/local/lib/python3.9/dist-packages/ansible/inventory/manager.py", line 38, in <module>
    from ansible.plugins.loader import inventory_loader
  File "/usr/local/lib/python3.9/dist-packages/ansible/plugins/loader.py", line 26, in <module>
    from ansible.utils.collection_loader import AnsibleCollectionLoader, AnsibleFlatMapLoader, is_collection_ref
ImportError: cannot import name 'AnsibleCollectionLoader' from 'ansible.utils.collection_loader' (/usr/local/lib/python3.9/dist-packages/ansible/utils/collection_loader/__init__.py)
+ cd /home/pi/screenly/ansible
/dev/fd/63: line 174: cd: /home/pi/screenly/ansible: No such file or directory
+ sudo -E ansible-playbook site.yml --skip-tags system-upgrade
ERROR! Unexpected Exception, this is probably a bug: cannot import name 'AnsibleCollectionLoader' from 'ansible.utils.collection_loader' (/usr/local/lib/python3.9/dist-packages/ansible/utils/collection_loader/__init__.py)
the full traceback was:

Traceback (most recent call last):
  File "/usr/local/bin/ansible-playbook", line 90, in <module>
    mycli = getattr(__import__("ansible.cli.%s" % sub, fromlist=[myclass]), myclass)
  File "/usr/local/lib/python3.9/dist-packages/ansible/cli/__init__.py", line 22, in <module>
    from ansible.inventory.manager import InventoryManager
  File "/usr/local/lib/python3.9/dist-packages/ansible/inventory/manager.py", line 38, in <module>
    from ansible.plugins.loader import inventory_loader
  File "/usr/local/lib/python3.9/dist-packages/ansible/plugins/loader.py", line 26, in <module>
    from ansible.utils.collection_loader import AnsibleCollectionLoader, AnsibleFlatMapLoader, is_collection_ref
ImportError: cannot import name 'AnsibleCollectionLoader' from 'ansible.utils.collection_loader' (/usr/local/lib/python3.9/dist-packages/ansible/utils/collection_loader/__init__.py)
+ /home/pi/screenly/bin/upgrade_containers.sh
/dev/fd/63: line 179: /home/pi/screenly/bin/upgrade_containers.sh: No such file or directory
+ sudo apt-get autoclean
Reading package lists... Done
```

MoJo2600, 2022-02-15:
Yes, it is working on raspbery pi os legacy

semaf, 2022-03-20:
> Use Raspberry Pi Os Legacy: https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/ https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy

Legacy Bullseye or Buster?

JonnyTech, 2022-03-20:
Legacy=Buster, Current=Bullseye .. Please follow and read the links before asking.

semaf, 2022-03-20:
Buster is not working with Raspberry Pi 4 8GB Rev 1.5

JonnyTech, 2022-03-20:
What is the issue exactly? Have just tried and boots correctly here.

ealmonte32, 2023-02-08:
This is what is open for new sprint release: https://github.com/Screenly/Anthias/projects/8

Also, this should not really be created as an "open issue", I think the open issues are getting out of control and most should be closed and not wait for the stale and then GitHub to automatically close.