Screenly/Anthias #1561: Screenly Won't Load Google Slides

lhuang-ast, 2021-11-09:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

I have Screenly OSE installed on a Rpi4 last week, and added a published Google slides as asset. It was working perfectly fine until this morning. If I add other websites to the asset list (including the original URL of the Google Slides), they show up fine. But when it's the published Google slide's turn, the screen just turns black.

I've tried re-adding the asset, re-imaging Rpi Lite, and re-adding Screenly OSE.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Publish a Google Slideshow, and use tinyurl to shorten the URL
* Enter the Asset URL 
* Press "On"
* Wait for the asset to show up on the screen

Help please. Thank you.

jerome-n, 2021-11-09:
Same issue here for my 2 Rpi. 
Others assets works but Google slide stay black. This configuration has worked for several years.
Any ideas ?

rtindol, 2021-11-09:
Same started today with our displays, Rpi3

willworkforicecream, 2021-11-09:
Same here on 3B+s. Here's the latest viewer logs from my test display. 

<html>
<body>

● screenly-viewer.service - Screenly Viewer
--
Loaded: loaded (/etc/systemd/system/screenly-viewer.service; enabled; vendor preset: enabled)
Active: active (running) since Tue 2021-11-09 11:45:57 MST; 57s ago
Process: 652 ExecStartPre=/usr/bin/xset s off (code=exited, status=0/SUCCESS)
Process: 725 ExecStartPre=/usr/bin/xset -dpms (code=exited, status=0/SUCCESS)
Process: 726 ExecStartPre=/usr/bin/xset s noblank (code=exited, status=0/SUCCESS)
Process: 728 ExecStartPost=/bin/rm -f /tmp/uzbl_* (code=exited, status=0/SUCCESS)
Process: 729 ExecStartPost=/bin/rm -f /tmp/screenly_html/* (code=exited, status=0/SUCCESS)
Main PID: 727 (python)
Tasks: 17 (limit: 2059)
CGroup: /system.slice/screenly-viewer.service
├─727 /usr/bin/python /home/pi/screenly/viewer.py
├─750 uzbl-core --config=- --uri=file:///tmp/screenly_html/black_page.html --print-events --connect-socket /home/pi/.cache/uzbl/event_daemon
└─755 /usr/bin/python2 /usr/bin/uzbl-event-manager -va start
 
Nov 09 11:45:59 raspberrypi python[727]: Generating asset-list...
Nov 09 11:45:59 raspberrypi python[727]: browser found dead, restarting
Nov 09 11:45:59 raspberrypi python[727]: Loading browser...
Nov 09 11:45:59 raspberrypi python[727]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 750.
Nov 09 11:46:01 raspberrypi python[727]: current url is file:///tmp/screenly_html/black_page.html
Nov 09 11:46:02 raspberrypi python[727]: Generating asset-list...
Nov 09 11:46:02 raspberrypi python[727]: Showing asset testscreen-large.jpg (image)
Nov 09 11:46:02 raspberrypi python[727]: Sleeping for 10
Nov 09 11:46:12 raspberrypi python[727]: Generating asset-list...
Nov 09 11:46:12 raspberrypi python[727]: Showing asset https://google.com (webpage)
Nov 09 11:46:12 raspberrypi python[727]: current url is https://google.com
Nov 09 11:46:12 raspberrypi python[727]: Sleeping for 10
Nov 09 11:46:33 raspberrypi python[727]: Generating asset-list...
Nov 09 11:46:34 raspberrypi python[727]: Showing asset GSlide Test Asset (webpage)
Nov 09 11:46:34 raspberrypi python[727]: current url is https://docs.google.com/presentation/d/e/2PACX-1vSF8HnYGol56VfS04VsKQZt6gpBSCfu09RhoD4AtK5SYTJZTbf1TAebNj8jWWVTenK6IXRkc1ZE7gpe/pub?start=true&amp;amp;amp;loop=true&amp;amp;amp;delayms=3000
Nov 09 11:46:34 raspberrypi python[727]: Sleeping for 10
Nov 09 11:46:44 raspberrypi python[727]: Generating asset-list...
Nov 09 11:46:44 raspberrypi python[727]: Showing asset Edited URL (webpage)
Nov 09 11:46:44 raspberrypi python[727]: current url is https://docs.google.com/presentation/d/e/2PACX-1vQE6hII-pGtPjg54aPfVjRXF2c9koL_r8tV7VAb4PzoySaJa5CT3i4sPlyBdha01tPO8Ej3LzdD1PqU/pub?
Nov 09 11:46:44 raspberrypi python[727]: Sleeping for 10

</body>
</html>

Local assets and other web assets that I've tried so far have worked, just not published google slides.

ImprobableConstruct, 2021-11-09:
I am having the same problem.  Google slides are just black but other assets,(pictures, websites, etc..) load just fine.  Is anybody looking into this?

lhuang-ast, 2021-11-10:
I'm guessing Google might have changed the coding of published sites, so the Screenly OSE browser doesn't support it anymore. I tried embedding the published slides to a Google site and load the asset. The Google site loads fine, but the embedded slides part just shows the loading circle thing. 

I don't know anything about coding to troubleshoot this, but hope this information can be of help.

pottsrichard, 2021-11-12:
This is really messy for us (a school with limited resources). We've used Slides instead of Sites, due to the 'can't use a mouse to click accept cookies' issue. All our content is now unavailable, other than a few emergency images we've uploaded. 
I know this has only just started, but is OSE maintained anymore? Things have been really quiet over the last year.

willworkforicecream, 2021-11-12:
If you are really committed to running the google slides method and can't
wait for a fix, you can check out FullScreenOS. It isn't as convenient,
feature rich, or reliable as Screenly OSE (which I am convinced is still
one of the best DigitalSignage solutions out there) but it can display a
google presentation. Just plug the published presentation URL in there and
schedule a cronjob to run the refresh script in the /home/pi/scripts/
directory.  Depending on your network setup, you may want to disable the
web server or restrict access with firewall rules since I don't think it
has any authentication.

pottsrichard, 2021-11-15:
> If you are really committed to running the google slides method and can't wait for a fix, you can check out FullScreenOS. It isn't as convenient, feature rich, or reliable as Screenly OSE (which I [...]
 
Thanks for the heads-up about FullScreenOS - I'll definitely take a look, as I can't see anything happening fast here. We originally had everything working with the Pi version of RiseVision, but were forced to switch once they started charging. I also once read something about a fork of Screenly OSE, but can't find the original link.

jerome-n, 2021-11-16:
If it helps, until the fix is released I converted my slides to movies with OBS. For us it is enough in the meantime.

timmy-tester, 2021-11-17:
We had the same google slides issue where google slides would just appear as black screen. After going to settings and upgrading to latest, it seemed to fix the issue.

willworkforicecream, 2021-11-17:
Also seems to be working again on my devices without any updates, so I guess Google fixed whatever they changed on their end.

closed 2021-11-17 by lhuang-ast