Screenly/Anthias #1562: Rasp4 No Splashscreen only comandline

flo258, 2021-11-11:
I have tryed all solutions and all fixes from ealmonte. No chance.

Has anyone an HowTo for the Raspi 4?

jfb74, 2021-11-12:
@flo258  I recently set mine up on my new Raspi 4 and did the following:
Based on the comment [here](https://github.com/Screenly/screenly-ose/issues/1355#issuecomment-696388611) I used the raspbian buster lite image and booted up into that. 

Then I used the latest install.sh from the master branch as so:
`bash <(curl -sL https://raw.githubusercontent.com/Screenly/screenly-ose/master/bin/install.sh)` I selected the Master/Development option (I believe it was #2) and then I answered No to all the questions. Once I did this it rebooted and was functional.

flo258, 2021-11-15:
I dont know why, i have tryed this several times with no success.
Now it worked. 

Thanks

closed 2021-11-15 by flo258