Screenly/Anthias #1564: Sprint 10 not booting

vatsake, 2021-12-23:
Raspberry pi 4 B does not boot from the Spring 10 image.
The LED is just lighting up - no activity.

shaqaruden, 2022-02-14:
Sprint 10 doesn't actually support RPI4 yet. Image SD card with latest RaspberryPi OS lite and install using the curl command in README.md. If you have any issues there tag me in your issues and I can assist you.

vatsake, 2022-02-16:
@shaqaruden I tried installing it with curl command.
It asked me to restart the system, but after that nothing happens?

It should open up a website with port 8080?.

kIrk2k, 2022-02-16:
@shaqaruden 
i installed via the curl command. and it works till the first reboot (pi4) after that the pi cant obtain an ip address anymore and cant start the webserver. Any ideas?

vatsake, 2022-02-16:
I tried again with Buster and all works well.

purebordem, 2022-04-03:
Reviving the dead a bit here, but using the RPI Imager software and choosing the Legacy OS Lite Buster doesn't appear to work with the curl command. Whole thing just runs through, choose reboot, and then just boots into console again.

vpetersson, 2022-04-04:
> Reviving the dead a bit here, but using the RPI Imager software and choosing the Legacy OS Lite Buster doesn't appear to work with the curl command. Whole thing just runs through, choose reboot, and then just boots into console again.
> 
> Ideas?

What exactly doesn't work?

purebordem, 2022-04-06:
It looks my issues were exactly what was being discussed in #1571, please disregard the previous comment